PX4/PX4-Autopilot #4861: Brake down sensor combined

tumbili, 2016-06-21:
As discussed with @bkueng 
We want to brake down the sensor combined topic into smaller topics, e.g. sensor_gyro, sensor_accel, sensor_baro. The reason for this are as follows:

1) Many modules are subscribing to sensor_combined but use just one value of the total 790 bytes, see e.g. navigator, fw_pos_control_l1, commander, ...
We can save a lot of memory if we split up the data.

2) Avoid expensive copying of data. Each orb_copy of the sensor combined topic results in 790 bytes being copied in memory.

3) Reduce latency from rates measurements to rate controller.

4) The topics contains lots of redundant data. The new logger system foresees to log the messages from the sensors directly so we don't need to have the sensor_combined topic.

We also would not need to publish three versions of the IMU data anymore since the voting of data could be handled in the sensor app. A special case is the ekf when we think about running three different instances based on three different IMU data sources. This case could still be handled by directly subscribing to the sensor data.

julianoes, 2016-06-21:
@tumbili I think this goes together with getting rid of sensors.cpp, and avoiding to copy all sensor data just for the sake of combining them into one struct, see #966.

tumbili, 2016-08-15:
This has been implemented.

closed 2016-08-15 by tumbili