PX4/PX4-Autopilot #4871: build system performance issues

dagar, 2016-06-22:
@LorenzMeier @bkueng @julianoes @tumbili 
Document your travis-ci, semaphoreci, etc issues here, especially performance related. I generally work on these things at night when they seem to be reasonably fast, but if the performance isn't acceptable during a typical development day document it here and I'll consider the options.

With the build environment containerized it would be easy to distribute the load between travis-ci, semaphoreci, circleci, and optionally jenkins.

I've also contacted travis-ci.

dagar, 2016-06-22:
Travis-ci OSX consistently has problems - https://www.traviscistatus.com/incidents/nf3tg08pqg2d

dagar, 2016-06-22:
@PX4/dev-team - Consider using your own PX4/Firmware fork for pull requests instead of pushing a new branch to the main PX4/Firmware repo. When you push a branch to the main repo travis-ci builds both the branch and the result of the merge. If you use a fork it only builds the result of the merge. This will help reduce our load on travis-ci and improve the build system for everyone. 
This has the side benefit of reducing branches off the main repo.

dagar, 2016-06-22:
@LorenzMeier could you help me with homebrew caching on travis-ci or put me in touch with another OSX developer?
The current OSX build spends the majority of its time on homebrew. Can you do a fetch into a travis-ci cached directory and offline install?

dagar, 2016-06-23:
Travis-ci support was helpful. I paid tier would give guaranteed concurrent jobs, but once we get a machine it's the same. I've stripped down the travis-ci build as much as I can.

dagar, 2016-06-25:
Travis-ci seems to be ~ 8-10 minutes per build now. If we want to reduce that any further (5 minute goal?) I think we'll need to start looking at our own hosting.

@LorenzMeier ping me if you see more problems or want to pursue other options.

closed 2016-06-25 by dagar

sanderux, 2016-06-25:
@dagar I have a qemu hypervisor (proxmox) in a hostingcenter 
do you have specs needed for running our own? or an image i could run as a test?

dagar, 2016-06-25:
Thanks @sanderux, but if you're paying per usage I suspect a virtualized compile machine is going to be quite expensive.

We have docker images for the dev environments (https://hub.docker.com/r/px4io/px4-dev-base/), otherwise it's just ubuntu + instructions (http://dev.px4.io/starting-installing-linux.html).

sanderux, 2016-06-26:
@dagar the hardware is mine, the only thing i pay for is data but i dont think that will exceed what i'm already paying for.
If i give you root to a vps, what environment would you prefer?

dagar, 2016-06-26:
Thanks Sander. Just stock Ubuntu 16.04 server, then I'd install jenkins and
docker. The web interface would need to be public to connect to github.

sanderux, 2016-06-26:
I'll pm you login details over skype.

sanderux, 2016-06-26:
How much disk, ram and cpu cores? I'd say 20g 4g and 4 cores? I have more than enough of all

dagar, 2016-06-26:
20GB disk is probably sufficient. The more memory and cpu you can throw at it the better. 4GB ram and 4 cores is roughly what travis-ci provides, although I suspect those cores are pretty heavily shared.

sanderux, 2016-06-26:
8gb/6cores setup. auth is in your mail.
let me know how you get on, if it works and improves performance we can set it permanently

dagar, 2016-06-26:
Thanks Sander!

sanderux, 2016-06-26:
i saw it quickly reached full mem when compiling, upped it to 10gig