PX4/PX4-Autopilot #4872: Estimator stability on fixed wing in auto

LorenzMeier, 2016-06-22:
We currently seem to have stability issues in master on fixed wing in auto mode.

LorenzMeier, 2016-06-22:
@tumbili @bkueng Found this nasty load issue in sensors:

```
Processes: 23 total, 6 running, 17 sleeping
CPU usage: 99.98% tasks, 0.02% sched, 
DMA Memory: 5120 total, 2560 used 3072 peak
Uptime: 3300.474s total, 924.324s idle

 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
   0 Idle Task                  924324  0.000     0/    0   0 (  0)  READY 
   1 hpwork                      87691  2.036   772/ 1592 192 (192)  w:sig 
   2 lpwork                       8647  0.000   572/ 1592  50 ( 50)  READY 
   3 init                         1518  0.000  1272/ 2496 100 (100)  w:sem 
 300 sdlog2_writer               55963  0.000  1044/ 2040 100 ( 95)  w:sem 
 327 top                            58  0.285  1212/ 1696 100 (100)  RUN   
 136 dataman                        15  0.000   684/ 1192  90 ( 90)  READY 
 156 sensors                    335626 56.966  1620/ 2192 250 (250)  w:sem 
 158 commander                   54656  0.686  3300/ 3592 140 (140)  w:sig 
 159 commander_low_prio           5811  0.000  2692/ 2992  50 ( 50)  READY 
 169 px4io                      170360 17.959  1028/ 1392 240 (240)  w:sem 
 175 mavlink_if0                 26711  0.022  2404/ 2792 100 (100)  w:sem 
 176 mavlink_rcv_if0              2648  0.240  1556/ 2096 175 (175)  w:sem 
 183 mavlink_if1                 20414  0.102  2404/ 2792 100 (100)  READY 
 184 mavlink_rcv_if1              3528  0.022  1140/ 2096 175 (175)  w:sem 
 324 gps                           353  0.000   672/ 1192 220 (220)  w:sig 
 211 mavlink_if2                114298  0.102  2404/ 2792 100 (100)  READY 
 212 mavlink_rcv_if2             31992  0.091  1564/ 2096 175 (175)  w:sig 
 219 sdlog2                     253345  0.812  2892/ 3392 177 (177)  w:sem 
 253 ekf_att_pos_estimator      925236 10.283  3460/ 4592 235 (235)  w:sem 
 261 fw_att_control             135390  3.992   880/ 1392 250 (250)  w:sem 
 266 fw_pos_ctrl_l1              74687  6.302  1228/ 1392 250 (250)  w:sem 
 269 navigator                   65056  0.000  1012/ 1296 105 (105)  w:sem 
```

LorenzMeier, 2016-06-22:
To repro:

```
gps stop
gps start -f
commander mode auto:loiter
commander arm
```

LorenzMeier, 2016-06-22:
Autostart: Caipirinha (3100)

julianoes, 2016-06-23:
Is this something about the estimator itself or just sensors doing something weird?

tumbili, 2016-06-23:
Fixed in #4884

closed 2016-06-23 by tumbili

julianoes, 2016-06-23:
@tumbili Do you mind closing issues once a fix is merged and not when a PR has been created. I think github mostly does it for you anyway.