PX4/PX4-Autopilot #4876: NAV_DLL_ACT value meaning inconsistent between PX4 & QGC

chobitsfan, 2016-06-23:
In QGC, NAV_DLL_ACT has several value option: loiter, RTL, land at current position. But modules/commander/commander.cpp only care about if NAV_DLL_ACT(datalink_loss_enabled) > 0

And NAV_DLL_ACT do not work in most modes except MAIN_STATE_AUTO_MISSION
according to set_nav_state() in  modules/commander/state_machine_helper.cpp
only when internal_state->main_state == MAIN_STATE_AUTO_MISSION, it will handle data_link_loss_enabled
in other modes like MAIN_STATE_POSCTL or MAIN_STATE_ALTCTL, it do not handle data_link_loss_enabled

julianoes, 2016-06-23:
Did you verify that the behavior is not as expected using e.g. SITL? There is also some logic in the navigator, it's just unfortunate that you don't see it by grepping.

Have a look at:
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/navigator_main.cpp#L158
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/navigator_main.cpp#L575

chobitsfan, 2016-06-27:
I do verify it in SITL/jmavsim.  I switch simulated drone to POSCTL and take it off using QGC & joystick.  Then I kill QGC and found drone just hover there (or keep going up if I kill QGC while take off). But drone will auto land if I put drone in mission:pause mode and then kill QGC

julianoes, 2016-07-06:
@chobitsfan Looking at the code I would expect it to follow the command after 10s. Can you check if the datalink loss is successfully detected in the commander?

julianoes, 2016-07-07:
@chobitsfan: I just verified this and you're right. 
However, The question is just what the sense is of a datalink lost in a manual mode such as Manual/Altitude control/Position control. In any of these modes you can still fly using the joystick or RC input. If you lose RC in that case, it will RTL anyway. The reason it does RTL (or something else) in auto modes is because the auto modes are allowed if there is on RC signal.

Does that make sense? Or what is your particular use case?

chobitsfan, 2016-07-07:
Thank you very much for your reply (I am sorry for my poor English)
This is my real case
I connect raspberry pi to pixhawk (through telemetry 1 port). My laptop connect to raspberry pi through wifi. I use QGC & joystick to fly quadcopter in POSCTL mode. I fork a mavproxy in raspberry pi to relay pixhawk mavlink to QGC. Everything works fine. But during another test flight. I move quadcopter forward and suddenly I saw raspberry pi drop from quadcopter (my fault, seems I did not attach it properly). In this case data link lost (wire between raspberry pi & pixhawk is gone). But my quadcopter keep moving forward until it hit a tree.

Maybe I shall not control quadcopter in this manner? ( I did not attach RC to pixhawk)

LorenzMeier, 2016-07-07:
@julianoes The resolution seems to be to trigger on data link lost in posctrl if RC was never present and data was present and now lost.

julianoes, 2016-07-07:
Ok, but should we ever be in POSCTL without RC?

dagar, 2016-07-07:
Yes, if you have manual control from QGC and a joystick.

julianoes, 2016-07-07:
Isn't that sent over RC_... or RC_...OVERRIDE?

dagar, 2016-07-07:
I believe it's MANUAL_CONTROL unless something's changed.
http://mavlink.org/messages/common#MANUAL_CONTROL

chobitsfan, 2016-07-08:
Yes,I found QGC send MANUAL_CONTROL while I use logitech joystick

julianoes, 2016-07-08:
I see. That means we need to either handle MANUAL_CONTROL lost as RC lost, or as @chobitsfan suggests, add the datalink lost failsafe to Manual/ALTCTL/POSCTL. I vote for the latter.

LorenzMeier, 2016-07-08:
Are you around for an in-person discussion in an hour? Would be easier.

gregd72002, 2016-10-09:
Hi, I experience the same problem when using Joystick only:
- arm
- set  manual mode
- switch off/close qgc
and the motors will keep spinning. 

What I was expecting is data link loss to occur and the copter to land at current position after 10sec. This never happened.

I would be expecting that MANUAL_CONTROL loss is the same as RC loss.

julianoes, 2016-10-10:
@gregd72002 thanks for the confirmation.

nanthony21, 2016-10-11:
Not sure if this is relevant but when using virtual joystick through QGC with the commander parameter set to "no RC checks" I get constant "manual control lost"/"manual control regained" messages. Looking through the code it looks like maybe the RC check section responsible for these warnings should be bypassed via status_flags.rc_input_blocked but it is not.

TSC21, 2017-08-20:
Stale

Mocart96, 2017-11-12:
Some time ago I moved to PX4 from ardupilot firmware due 2 functions: virtual joys and "data link lost". My copert comunicates with QGC using LTE and failsafe tigger base on data lost is critical. Is there any clear description how "data link lost" tigger works ? As I understand "RC lost" tigger works in manual modes and if the case eg QGC is disabled it can change mode to eg. land only. Will this "RC link" trigger work also for "mission" mode or should I set up "data link lost" for mission mode only. Thanks in advice for any additional clarifications.

dagar, 2017-11-12:
@Mocart96 
 - RC lost is based on the last manual input. This could be real RC or a gamepad over mavlink
 - data link lost is based on the HEARTBEAT in a mavlink telemetry stream

These become flags used by the state machine in commander.

For MISSION loss of RC will only trigger a failsafe event if you don't have data link loss configured.

I'm looking for ways to generate documentation for the state machine. Too often the best answer for some of these more specific questions is to check the code.

Mocart96, 2017-11-12:
@dagar 
Daniel, Lets me comment Virtual Joystick option first.  I`m using UDP link over LTE to communicate copter with QGC. 

> RC lost is based on the last manual input. This could be real RC or a gamepad over mavlink.

When I setup “COM_RC_IN_MODE” parameter as “Joystick/no RC check” then virtual joysticks are always available on screen and it works like regular RC controller when LTE link is alive. 
Any action that disturbing/killing this connection (like: drop on LTE signal or crash/close of QGC software) will force copter to start failsafe action if “RC Loss Failsafe Tigger” defined.  It should works for manual and mission modes. Correct me if I`m wrong. 

> data link lost is based on the last HEARTBEAT in a mavlink telemetry stream from the GCS.
> For MISSION loss of RC will only trigger a failsafe event if you don't have data link loss configured.

Base on your comment I should have the same behaviour for “Mission” mode if  “data link loss” parameter is disabled. So my question is when should I use “data link loss” trigger in this case? 
I was thinking before that this trigger works any time when eg. UDP link is down. If there is no link and no HEARTBEAT between QGC and Flying controller than failsafe can enabled by using this trigger. 
But seems it works different because it`s not available for manual modes.
Sorry for too many words but I trying to mitigate risk of using my copter without RC controller and I cannot find overall description how it works. I not coder, It hard to me to understand the code.
Let me know if you have any suggestion how to setup the most safe failsafe options for no RC flights if possible. What exactly can give me enabling “data link loss” trigger with no rc setup?

LorenzMeier, 2019-03-15:
@julianoes looks like @NAmmann has a similar issue that he just raised on Slack.

NAmmann, 2019-06-25:
Any updates on this issue?

chobitsfan, 2019-06-25:
I am sorry but I am working on other matters. Maybe I should close it?

julianoes, 2019-07-01:
This needs further testing for work on commander.