PX4/PX4-Autopilot #4878: LPE: LocalPos-Z not accurate with latest changes

ecmnet, 2016-06-23:
@jgoppert With latest master something with LPE must have changed massively. I've SF10A Lidar and PX4Flow attached. Sonar altitude seems not to be recognized at all, and even with LIDAR the estimated altitude has an accuracy of +/-20cm, which is far beyond the LIDARs resolution. I also tried to adjust LPE params (e.g. setting bro std.dev. to 3m). In former LPE implementations I received an accuracy of +/-2 cm, with sonar data available even better. Any ideas?

ecmnet, 2016-06-23:
Might be related to https://github.com/PX4/Firmware/issues/4841

jgoppert, 2016-06-23:
What do you have mpc alt mode set to?

ecmnet, 2016-06-23:
It is set to 1

jgoppert, 2016-06-23:
Log?

ecmnet, 2016-06-23:
No, I had only I short flight indoors with those altitude changes. Meanwhile I deleted the logs unfortunately. But leaving the vehicle still in an altitude of 40cm shows that inaccuracy. Raw distance is measured properly by the LIDAR. I am also curious that no sonar messages (init, etc) appear - also no LIDAR init, but for LIDAR I receive fault messages if setting the std.dev. below 0.03. I also remember that I had set bro std.dev. to 5m earlier. This is now not possible anymore (limit is 3).

ecmnet, 2016-06-23:
Try to return to an earlier build now

ecmnet, 2016-06-23:
Seems to be related to the driver SF10a

jgoppert, 2016-06-23:
If you can get a log of it running on your table at 40 cm ,that would be interesting.

ecmnet, 2016-06-24:
Here is a log for an altitude of 55cm: http://logs.uaventure.com/view/ujQ97RjkN9xVczowfJA5W3.
Some observations: setting LPE_BAR_Z to 5m significantly improves accuracy, but looking at the RAW_DISTANCE topic published, there are a lot of spikes, which I am not aware of with SF10a (there are no spikes at all with it besides +/-1cm jumps according to its accuracy).

ecmnet, 2016-06-24:
[image]

Comparison raw data: Sonar (top, green) vs. LIDAR (bottom, green)

jgoppert, 2016-06-24:
Log muncher doesn't plot agl yet and that is being regulated not local z.
Local z is a best estimate of asl relative to home, so it contains baro
drift. Just making sure you realize that. Need to plot agl = ( asl vehicle
- asl terrain) = -( est0.s2  - est0.s9)

ecmnet, 2016-06-24:
Ok. Understand that. What is est0.s2, est0.s9? Nevertheless, even when setting Baro Std.Dev. to 30 the drift occurs (I assumed, that in this case baro should not be considered so much). Here an comparison (BaroStd.Dev=3) between LPE-Z and LIDAR raw distance:

[image]

Please note also, that in the past I did not have such a drift with LPE.

jgoppert, 2016-06-24:
Take a look at the code. The difference is just that I'm publishing local z using the estimator z state instead of agl like I used to.  So you can't compare them. Try plotting dist bottom.

ecmnet, 2016-06-24:
What do you mean by bottom-distance - altitude-relative? Even then - why are the relative changes occurring?

Nevertheless, I think this issue is not about choosing the correct key figure, as I had those Z instabilities  also during flight as mentioned above.

jgoppert, 2016-06-24:
I'll take a detailed look at your log tonight. Those fields are log names.

ecmnet, 2016-06-24:
Yes, i figuered that out already. The bottom clearance should be the distance to ground. I also noticed a significant offset between sonar and lidar. I tweaked the params to compensate this and will give it a flight tomorrow. Thanks for helping so far.

ecmnet, 2016-06-25:
@jgoppert Here is the latest log with altHold working quite well: http://logs.uaventure.com/view/sPBWayzBVCdtE6xYfMNzYJ. I am now quite curious, what it has been before. Could be that is was the lidar-spike-issue (caused by WLAN to close to lidar). So sorry for causing any inconvenience here. Mainly it was caused by the fact that I was not aware of the LPOS-Z definition change in LOCAL_POSITION_NED. I am closing this for now.

ecmnet, 2016-06-25:
@jgoppert Wouldn't it make sense to publish LOCAL_POSITION_NED_TARGET for LPOS-Z in the same way as actual? Currently those two definitions do not fit.

jgoppert, 2016-06-25:
The estimator doesn't publish that. If mpc alt mode is set to 1, it should
treat that as agl ( altitude above ground level). I agree it is confusing.
Maybe if mpc alt mode = 1, then we publish agl as local z like we used to.

ecmnet, 2016-06-25:
Would be great if target and actual would have the same definition, regardless what mpc alt mode is set to.

LorenzMeier, 2016-06-25:
We're not going to mess with the local Z value. I don't think we have to argue about the properties of a rigid body transformation? I'm watching this discussion with horror - please fix this up to use the field which is present for this use case, and please make sure to use that in the controller:
https://github.com/PX4/Firmware/blob/master/msg/vehicle_local_position.msg#L30

AGL is no coordinate system. Please stop trying to use it like one.

mhkabir, 2016-06-25:
@LorenzMeier It's already done properly. The estimator publishes AGL to the field you mentioned, and the controller uses it in case the parameter is set. 

The main question (and what we have to actually fix) here is, how do we reconcile the logging of the target and current altitudes when the controller is in terrain following mode. Because currently, in terrain follow mode, the target alt is displayed in AGL, and the actual alt is displayed on Local NED.

LorenzMeier, 2016-06-25:
The target alt should be transformed given the current AGL and used in the control loop and logged that way. The controller param should just enable or disable the application of the AGL offset.

jgoppert, 2016-06-25:
If we transform the target altitude using agl, the setpoint will be moving due to baro drift with no user input, which isn't horrible, if that is the direction we want to go. The alternative is to just use dist bottom in log muncher so users can look at alternate plot when in terrain following mode.

ecmnet, 2016-06-25:
I agree, WE should not mess up any definitions, but we should be consistent at least.

jgoppert, 2016-07-01:
@ecmnet If you have time can you see if this is resolved on master now?

ecmnet, 2016-07-03:
@jgoppert hopefully I'll  manage to confirm this by tomorrow. Could you summarize the changes ups to now so far?

jgoppert, 2016-07-03:
Even when in altitude following mode instead of terrain following mode, the behavior should be more reasonable. As part of testing the racing drones, we realized that the baro noise should be increased.

We really still need to address recording the moving altitude set point correctly.

ecmnet, 2016-07-04:
@jgoppert I am now again facing those inaccuracies resulting in 'jumps' at altitude hold: http://logs.uaventure.com/view/eZf7XjBAxmRHRYQWxDjukW. Maybe you could have a look?

jgoppert, 2016-07-04:
The main problem is that your terrain estimate was drifting off while you were on the ground and your lidar was out of range. Try setting LPE_EST_TZ = 1e-5 to lock it down.

When my lidar is on the ground it reads 30 cm, which is the min value. I'm not sure why yours is reading 0, that might be the root cause of this.

[image]

Because terrain was then incorrect when lidar started reading, it caused lidar to fault so that lidar wasn't being used when you were in the air:

[image]

[image]

jgoppert, 2016-07-04:
@ecmnet Just found why terrain was drifting off, the sonar was reading, but the lidar was not on the ground. The sonar was reading very bad values, driving the terrain estimate to sink down below the floor since the baro said you were at the same altitude.

[image]

ecmnet, 2016-07-04:
@jgoppert Thanks for the quick analysis. According to its specification, SF10a is working from 0cm to 25m. I have also set LIDAR offset to -0.008. That the sonar was very faulty is clear, the vehicle had been over thick carpet. Might it make sense to set the lower limit of the LIDAR to 1cm in the driver?

jgoppert, 2016-07-04:
If it published at zero it should be fine. Maybe try increasing lpe Snr z to ignore sonar.

ecmnet, 2016-07-04:
@jgoppert I don't quite understand. If LIDAR was faulty, increasing SNR_Z would not help, I think. Alt. would then rely on baro only. For me the question remains, why LIDAR was considered faulty, not resuming at all.

ecmnet, 2016-07-04:
@jgoppert I now added 1cm minimum distance to the driver. As a result, LIDAR is initialized correctly each time when the vehicle rises. Still the altitude hold was not very stable. Here's the log: http://logs.uaventure.com/view/bce5sHWSAZLtEzpvWSru5k. Additionally: After switching to POSCtrl I faced a serious issue when flow was faulty. I had no chance to control the device anymore and it crashed.

jgoppert, 2016-07-04:
The 1cm seems to have done the trick for making it read correctly. Sorry to hear about your crash. Here is the analysis.

[image]

When flow drops out at 54 seconds, it keeps trying to estimate position for the next 6 seconds until the estimated std. dev. goes about the threshold of 1 m. At this point the position had drifted around 2 m. Then the light should have gone blue because it stopped publishing position. During this time it is basically in attitude control and you can see that you were controlling the attitude set points. Next, flow comes back and it is in position control briefly. The flow is then faulted for not agreeing enough with the estimate and the estimate change significantly, causing the vehicle attitude to react, causing the crash.

The best solution is probably to lower LPE_XY_PUB = 0.2 instead of 1. This will stop attempting to do position control when the estimated std. dev. is above 20 cm. This would have been more similar to the old method of giving up position control immediately after flow drops out.

jgoppert, 2016-07-04:
@ecmnet Do you know why the flow quality got so poor, I don't see this often. I documented my setup closely here: http://dev.px4.io/flow_lidar_setup.html if you haven seen it yet.

jgoppert, 2016-07-04:
Your altitude plots look reasonable. It is basically following lidar and sonar is ignored because it is faulting 90% of the time:

[image]

[image]

ecmnet, 2016-07-04:
@jgoppert That is quite normal in my surrounding - leaving the carpet there is no structure on the ground. I went into that area on purpose to check sonar readings. With previous versions of LPE (and of course also the rest of the firmware), I was able to steer back - even in PosHold mode, which was impossible today. Flow is setup correctly in working fine on structured ground.

Regarding the altitude, also your plots show an accuracy of about 10-15cm, far from what Lidar is capable to achieve. Also regarding this I had much better results in the past.

ecmnet, 2016-07-04:
@jgoppert Regarding the SF10a driver: Should the minimum boundary be implemented by default? What do you think?

jgoppert, 2016-07-04:
@ecmnet Yes, it needs a minimum. 0 is actually an error code on the sonar, so it throw 0's out. It would be ideal if you setup your sensor so that it was reading a valid number while on the ground, that would solve a lot of the odd behavior on takeoff.

I am going to try to fly with my lidar and try to reproduce your results and address the degradation in altitude hold. Do you have an old log you were happy with that we can compare against?

ecmnet, 2016-07-04:
@jgoppert The only I've kept is the one I mentioned above (March, 18th): http://logs.uaventure.com/view/RjyeEPwbDAgwgPePmYq2NL. Here baro was set to LPE_BAR_Z = 1.0. I remember, I got better results with higher baro values. In this log you can also see Flow working quite properly.

ecmnet, 2016-07-04:
@jgoppert Could it be that LIDAR timeout is detemined including the offset? It seems, that if I set the offset to 5cm (positive), a LIDAR timeout never occurs.

jgoppert, 2016-07-06:
@ecmnet Yes, the timeout does include the offset.

```c++
	// measure
	float d = _sub_lidar->get().current_distance + _lidar_z_offset.get();
	warnx("d %10.2g, lidar z offset %10.2g\n", double(d), double(_lidar_z_offset.get()));
	float eps = 0.01f;
	float min_dist = _sub_lidar->get().min_distance + eps;
	float max_dist = _sub_lidar->get().max_distance - eps;

	// check for bad data
	if (d > max_dist || d < min_dist) {
		
```

If your difference is 3 cm, it is probably best to include the offset from the c.g. for both. For instance:

If the lidar were 5 cm below the c.g. of the vehicle, set it  LPE_LDF_OFF_Z = 0.05.
If the sonar were 8 cm below the c.g. of the vehicle, set it to LPE_LDR_OFF_Z = 0.08

jgoppert, 2016-07-06:
@ecmnet I think the offset should not go into the timeout calc, I will change that. For instance, if the users sets the offset to 1, then even when the lidar is reading 0, it will read as 1 and be considered valid.

ecmnet, 2016-07-06:
@jgoppert Thanks for that, I think it make sense to ignore the offsets for timeout identification. My previous setup was exactly as you described above. I'll do some testflights tomorrow with the latest changes you applied.

ecmnet, 2016-07-07:
@jgoppert Here the log of a short test-flight with Altitude hold. Still 10 to 15 cm variance in altitude.:
http://logs.uaventure.com/view/7tFtYqqcdW53LmtvUFc4nd

ecmnet, 2016-07-10:
@jgoppert Another test flight failed (same environment, default parameters). LIDAR was initialized during manual takeoff (variance > 10cm) and faulted immediately after that. Switching to althold without noticing that caused althold rely on baro only.

jgoppert, 2016-07-10:
@ecmnet Do you have a log of your last flight?

ecmnet, 2016-07-10:
I just updated my message: http://logs.uaventure.com/view/JVCgUgiNE7hg2MFKDSEHJB

jgoppert, 2016-07-11:
@ecmnet  How far is the physical lidar sensor off of the ground? It seems odd that it is still reporting 0 before the flight. Also, the sonar LPE_SNR_Z of 5 cm is causing the estimate covariance to converge which makes it more likely to throw out lidar if it deviates.

[image]

This is a plot of the faults.

[image]

In this plot you can see that the raw lidar without offset is reporting 0 when on the ground.

[image]

In this plot you can see that when the sonar isn't faulted, it is driving the estimator covariance down.

ecmnet, 2016-07-11:
lidar is about 4cm above ground. I figured out that it gives valid readings from about 5cm on.

jgoppert, 2016-07-11:
If you are unable to mount the lidar so it can read well in this environment, I think the only solution is to increase LPE_SNR_Z so that the estimate never gets too sure of it self while sitting on the ground.

ecmnet, 2016-07-11:
Why is the sonar taken into account when sitting on the ground? As I understand, it's range is starting at 30 cm.

ecmnet, 2016-07-11:
Another topic is, that altitude terrain and bottom clearance are only published, when Flow is valid. I assume that is, because I've switched of GPS and global position is based on flow only now. Altitude terrain and bottom clearance seem to depend on the global position.

ecmnet, 2016-07-11:
@jgoppert Another test: http://logs.uaventure.com/view/vYKNzU4uxSkbYt7yVxd7Bh. LPE_SNR_Z was now set to 0.3. Additionally I set the min-value of the lidar to 0.05m. Altitude seemed to be relied on baro mostly now.

jgoppert, 2016-07-11:
The issue is that sometimes the sonar will report values as high as 300 cm while on the ground (see plot below). I use fault detection to throw those out, but eventually it causes some drift in the estimate.

In this plot you can see the point where the estimator chooses sonar over lidar. Sonar has a std. dev. of 5cm nominally and lidar has a std. dev. of 3 cm, so when sonar is closer, it starts faulting lidar. I'm checking the math to make sure that fault threshold is accurate, because to fault lidar on 10 cm seems a bit high.

[image]

jgoppert, 2016-07-11:
@ecmnet Here is a plot of your last flight. The estimate if following lidar exactly as long as it is present, so it isn't baro's fault

[image]

Also, If you look at the terrain state, it has drifted 3 m. If you are flying indoors with level ground, you can set LPE_T_Z = 0.001 or less. The default of LPE_T_Z = 0.1 is for outdoor/ hilly environments.

jgoppert, 2016-07-11:
Here is a zoom on the lidar and agl/ smoothed agl that is used for control:

[image]

ecmnet, 2016-07-11:
At about sec 18, I switched to AltHold - and it rose to 0.7m. That's a bit weird ( and after my last indoor crash I am a bit scared..)

[29 more replies]

closed 2017-04-11 by dagar