PX4/PX4-Autopilot #4879: mv_att_control integral control not working

ndepal, 2016-06-23:
I've noticed that integral control on my quad (PixFalcon flashed with latest master px4fmu-v2_default) is not working properly. This is independent of parameter tuning.

I've tracked down the problem to [this line in the attitude controller](https://github.com/PX4/Firmware/blob/master/src/modules/mc_att_control/mc_att_control_main.cpp#L782). The motor limits are always nonzero, preventing the controller from integrating.

I notice that in SITL, the motor limits are zero.

What is the intended behavior of the motor limit values? If they are supposed to be 0 in general, what could be causing them to be nonzero for me?

LorenzMeier, 2016-06-23:
@tumbili Can you have a look?

tumbili, 2016-06-24:
@ndepal Could you please attach a log file?

ndepal, 2016-06-24:
Here is a log file: http://logs.uaventure.com/view/sguauUpKefkZXdCU3LB377.

I'm hovering in manual. You can see that there is a constant offset between the pitch and the pitch setpoint. I'm having to push the pitch stick forward constantly in order to achieve hover.

tumbili, 2016-06-27:
@ndepal @LorenzMeier Fixed by #4922

closed 2016-06-27 by tumbili