PX4/PX4-Autopilot #4899: EKF2 heading and altitude oscilations

svpcom, 2016-06-24:
I've found that latest stable and beta branches in ekf2 mode (make px4fmu-v2_ekf2) has
strange oscillations when copter is standing on the ground (armed, but without props).  

ekf2 estimator (beta) constant heading drift 
http://logs.uaventure.com/view/Jp6CwPbKLKmEB3ToAaEoTn

ekf2 estimator (stable) altitude oscillations
http://logs.uaventure.com/view/jLxcdP8Qu3RQdvs5chsCre

inav estimator (stable) no any oscillations and drift
http://logs.uaventure.com/view/u2MGrNLNmcWPht3Jv4DVAD

copter conf: hexa in X mode, no lidar, no sonar, pixhawk with external compass and gps (ublox m8n)
I've calibrated copter several times to ensure that is not a calibration problem.

svpcom, 2016-06-24:
All three measurements was made in the same place (clear sky, 18 gps+glonass sats visible) in 10min interval

tumbili, 2016-06-24:
@svpcom It looks like you are using a pretty old version of ekf2. I noticed that this version still has the gyro scale estimation which is not in the new version anymore.

What you are seeing are no oscillations but state resets. The reason is that your accel sensor data is off by much. The magnitude of the acceleration vector when the vehicle is at rest is only about -8.8 instead of -9.81 m/s^2. I therefore assume that you are having some king of calibration or sensor problem. One thing to look for is what the accel bias estimation did, however, as I mentioned this version is deprecated and we are not going to look into it.

Can you try to find out what is wrong with your sensor data and then give PX4 master a try?

closed 2016-06-25 by tumbili

svpcom, 2016-06-26:
@tumbili I've recalibrated sensors using level tool and use latest master - it helps.
But i've found a strange behavior - with uncalibrated accelerometer ekf2 estimator (from latest master branch) ground station show me moving on the map with ~ 500m/s (!!) horizontal speed (copter stay on the ground). GPS is running, has 3d fix (13 sats with clear sky). Is it intended behavior (accel integration overrides gps position)?

tumbili, 2016-06-27:
@svpcom Sounds like something is very wrong. Can you arm the copter and post a log of that?

svpcom, 2016-06-27:
@tumbili  I is very hard because with uncalibrated sensors i'm unable to arm (and start logging). 
Steps for reproduce:
1. Reset to factory defaults. (all sensors become uncalibrated)
2. Load config [conf.zip](https://github.com/PX4/Firmware/files/335901/conf.zip)
3. Calibrate gyro and level horizon.
4. Go outdoor and start pixhawk with attached gps.
5. Wait for gps fix + several minutes
6. See a drift