PX4/PX4-Autopilot #4908: Snapdragon: replay log has drops

julianoes, 2016-06-25:
Recent flight testing with Snappy yielded replay logs with drops/timeouts. As far as I know this was working (better) before and therefore needs git bisecting.

Simonnnnnnn, 2016-06-27:
ur snappy already can fly ?

julianoes, 2016-07-05:
@Simonnnnnnn yes, sure can :)

julianoes, 2016-07-05:
It looks like we're not looking at logging drops as such but rather timing issues. Needs more investigation.

julianoes, 2016-07-05:
On a closer look I can see that sdlog2 which runs on the Linux side sometimes misses updates from the (m)uORB.

LorenzMeier, 2016-07-05:
Can you raise that with @mcharleb and look at the total load of the px4 process?

If it's significantly less than 100% it hints at an issue with muorb.

julianoes, 2016-07-05:
@LorenzMeier I've already checked the load of the mainapp process and it was around 30%. I'm gonna try to debug it with some simple plain subscriber to see if I can reproduce it.

julianoes, 2016-07-05:
Strangely, htop shows that one core is used 30% while the mainapp consumes around 100%.

julianoes, 2016-07-05:
I wrote a simple subscriber that uses poll. It never misses a replay message (except on startup) while sdlog2 keeps missing one or two every now and then, maybe around twice a minute.

julianoes, 2016-07-05:
The issue doesn't really seem to be CPU load dependant as it also happens if only sdlog2 (and uorb, muorb) is running on the Linux side.

LorenzMeier, 2016-07-05:
@julianoes Make sure nothing is skipped here:
https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2.c#L1477-L1483

This should not happen in replay or all logging mode, it is expected to happen in normal logging mode.

LorenzMeier, 2016-07-05:
Also make sure that the priority is set correctly - if unsure set to max:
https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2.c#L328-L343

LorenzMeier, 2016-07-05:
@bkueng What is the state on the new logger for EKF2 replay?

bkueng, 2016-07-06:
It works once #4917 and #4817 get merged.

julianoes, 2016-07-06:
> @julianoes Make sure nothing is skipped here:
https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2.c#L1477-L1483

Nothing skipped there.

> Also make sure that the priority is set correctly - if unsure set to max:
https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2.c#L328-L343

`SCHED_PRIORITY_MAX` fixes it but that's probably a bad idea for logging. I'm wondering what could cause the drops.

julianoes, 2016-07-06:
There are even drops with `SCHED_PRIORITY_MAX` just a lot less frequent.

LorenzMeier, 2016-07-06:
On Snapdragon running the logger on Linux at high prio is fine. This is just something that should not be done on the DSP / STM32. I find it still confusing that  you see drops. That again means going back into muORB.

julianoes, 2016-07-06:
It is confusing, there could be several things that are influencing this.

E.g. I see a lot less missed updates on my bench setup which doesn't have a external GPS/mag connected.

julianoes, 2016-07-06:
There must also be some power saving going on, at least on the Linux side. A simple `usleep(1000)` loop  leads to waits > 5000 at least once a second without high CPU load. Once I run `stress -c 4`, everything is running as expected.

LorenzMeier, 2016-07-06:
That makes sense. We even see that on i7 boards, I just forgot, sorry. Can you work out the best way to force the right operation of the core and potentially also CPU affinity and get this going?

julianoes, 2016-07-13:
Considerably improved with #5003. There are still some drops every now and then but that should be bearable. Closing for now.

closed 2016-07-13 by julianoes