PX4/PX4-Autopilot #4909: LPE GPS timeout

svpcom, 2016-06-25:
I had a test flight today with px4 stable branch + LPE estimator.
Several times there were a strange issues when copter in POSCTL mode becomes drift very fast.
I've inspect a logs and found following messages:
```
2016_06_25_10_13_10: [lpe] gps init lat  55.23 lon  37.74 alt 186.3 m
2016_06_25_10_13_10: [lpe] xy resume 
2016_06_25_10_13_10: [lpe] home lat  55.23 lon  37.74 alt 119.2 m
2016_06_25_10_13_11: [blackbox] /fs/microsd/log/2016-06-25
2016_06_25_10_13_11: [blackbox] recording: 13_13_10.px4log
2016_06_25_10_13_12: [lpe] home lat  55.23 lon  37.74 alt 119.2 m
2016_06_25_10_13_14: LANDING DETECTED
2016_06_25_10_13_26: TAKEOFF DETECTED
2016_06_25_10_13_26: [lpe] home lat  55.23 lon  37.74 alt 119.2 m
2016_06_25_10_13_28: LANDING DETECTED
2016_06_25_10_13_28: TAKEOFF DETECTED
2016_06_25_10_13_29: [lpe] home lat  55.23 lon  37.74 alt 119.2 m
2016_06_25_10_13_41: [lpe] GPS timeout 
2016_06_25_10_13_41: [lpe] xy timeout 
2016_06_25_10_13_48: REJECT AUTO RTL
2016_06_25_10_13_54: [lpe] gps init lat  55.23 lon  37.74 alt 195.6 m
2016_06_25_10_13_54: [lpe] xy resume 
2016_06_25_10_15_00: LOW BATTERY, RETURN TO LAND ADVISED
2016_06_25_10_15_03: LANDING DETECTED
2016_06_25_10_15_03: DISARMED by RC
2016_06_25_10_15_04: [blackbox] stopped (3611 drops)
2016_06_25_10_15_06: [lpe] GPS timeout 
2016_06_25_10_15_06: [lpe] xy timeout 
2016_06_25_10_15_38: ARMED by RC
2016_06_25_10_15_39: [blackbox] /fs/microsd/log/2016-06-25
2016_06_25_10_15_39: [blackbox] recording: 13_15_38.px4log
2016_06_25_10_15_41: DISARMED by RC
2016_06_25_10_15_42: [blackbox] stopped (180 drops)
2016_06_25_10_15_45: ARMED by RC
2016_06_25_10_15_46: [blackbox] /fs/microsd/log/2016-06-25
2016_06_25_10_15_46: [blackbox] recording: 13_15_45.px4log
2016_06_25_10_15_52: DISARMED by RC
2016_06_25_10_15_54: [blackbox] stopped (0 drops)
``` 

http://logs.uaventure.com/view/RZYxAMx9kWHBMCVkTc3kB9

After switching estimator to *inav*  gps timeout issue disappears and there were no any drifts.

LorenzMeier, 2016-06-25:
@jgoppert I had the same on my test system.

jgoppert, 2016-06-25:
The gps signal is poor so it  stops publishing position and effectively drops back to altitude mode. To fix we need to lower gps requirements.

jgoppert, 2016-06-25:
Will look to find what gps  requirement failed in your log asap.

jgoppert, 2016-06-26:
Here is why it timed out, your EPH went above 3.

[image]

Also, your version is somewhat outdated, as you don't have LPE_EPV_MAX (parameter for vertical that was reacently added)

jgoppert, 2016-06-26:
So here the options:

* Increase defaut LPE_EPH_MAX
* When we exceed LPE_EPH_MAX after boot, do not stop using GPS to estimate position.

Also I like this idea:

* Make estimator position timeout based on covariance of estimate instead of a fixed time interval, but still 5 seconds above the max EPH is still probably not going to work, so we will need to handle that with one of the other two steps mentioned above.

iZhangHui, 2016-06-28:
Optical flow has the same problem.

nsh> INFO  [local_position_estimator] [lpe] flow init: quality 153 std 19
INFO  [local_position_estimator] [lpe] xy resume 
INFO  [commander] home: 40.4300001, -86.9290010, 41.23
INFO  [local_position_estimator] [lpe] home lat  40.43 lon -86.93 alt  41.2 m
INFO  [local_position_estimator] [lpe] flow timeout 
INFO  [local_position_estimator] [lpe] xy timeout 
WARN  [navigator] navigator: global position timeout
INFO  [local_position_estimator] [lpe] flow init: quality 119 std 4
INFO  [local_position_estimator] [lpe] xy resume

mhkabir, 2016-06-29:
@iZhangHui You're likely flying over badly textured surfaces, where flow cannot stay locked. A log would help to confirm this.

jgoppert, 2016-06-29:
#4914 will improve this. It will attempt to navigate without flow longer.

PR #4914 (2016-07-01): LPE Variance Dependent Publication
LPE was relying on timeout (of last measurement with good stats) to govern when it should publish its estimates of position. However, this didn't work well for temporary errors in sensor stats that are resolved. For instance, #4909.

To handle this. I am making the timeouts be the time since a last valid measurement (not necessarily with good stats, such as EPH). However, the publication is now dependent on the estimator estimate of the state precision using the variance. This I hope will resolve #4909 but I still have to test it.
closed 2016-07-01 by jgoppert