PX4/PX4-Autopilot #4910: Support Pixhawk with 2MB flash (boards with rev3 chip)

mhkabir, 2016-06-25:
All the Pixhawks I have recently acquired have a rev 3 STM32 chip, and therefore should support the full 2MB.

Would make life much easier if we can add some sort of an auxiliary config to support these boards, and potentially @proficnc's Pixhawk 2s in a couple of months time, when they start retailing.

@davids5 @jgoppert

julianoes, 2016-06-25:
Is there any way to tell in software which revision we have (except triggering the known fault)?

dagar, 2016-06-25:
There's mcu_version, but I think you'll need a new config, px4fmu-v2_large or whatever you want to call it.
https://github.com/PX4/Firmware/blob/ce7dad81c2db0bcf319f62d3d9550f604a87ea1c/src/modules/systemlib/mcu_version.c#L116

Then you could add `cmake/configs/nuttx_px4fmu-v2_large.cmake`, maybe make it a superset of px4fmu-v2_default.  You'll either need a duplicated nuttx-config, or find a find to use a different ld.script. https://github.com/PX4/Firmware/blob/master/nuttx-configs/px4fmu-v2/scripts/ld.script

davids5, 2016-06-25:
Hi!

The latest revs of [bootloader] (https://github.com/PX4/Bootloader/blob/master/main_f4.c#L311) prevents that DL 2MB of a file to a "flawed device" - [see] (https://github.com/PX4/Bootloader/blob/master/main_f4.c#L578) 

So it is conceivable we could add a new target or attribute to the build to switch in a different linker file.
I do this in nuttx_v3 for bootloader. Or in the nuttx copy for that target 'sed' it.

The danger here is boards with old bootloader will allow the load - so I would add a module or ifdef ver  to check if the > then 1 meg is not 0xffffff and refuse to run,

proficnc, 2016-06-25:
This would be great!  
Also there are quite a few of the Pixhawk 1's that have the new chip.

kd0aij, 2016-11-08:
@davids5 I'm trying to figure out how to make a px4fmu-v3 config which differs from v2 only in the linker script. But I can't find anything similar in nuttx_v3; could you point me in the right direction?

davids5, 2016-11-08:
I think it is the same as you did for https://github.com/PX4/Firmware/pull/5821 but just use the /src/driver/boards/px4fmu-v2 in the cmake/configs/nuttx_px4fmu-v3_default.cmake

See https://github.com/PX4/Firmware/pull/5821/commits/43a7f83d1784f931d39bdadd96c92083a3e33d25

kd0aij, 2016-11-08:
It seems that include paths are constructed using the target name:
```
12:44:03 {add_fmuv3} ~/git/kd0aij/PX4/Firmware$ make px4fmu-v3_default 
-- nuttx-px4fmu-v3-default
-- CMAKE_INSTALL_PREFIX: 
-- cmake module path: /home/markw/git/kd0aij/PX4/Firmware/cmake
-- The ASM compiler identification is GNU
-- Found assembler: /usr/lib/ccache/arm-none-eabi-gcc
-- catkin DISABLED
-- Found PythonInterp: /usr/bin/python2.7 (found suitable version "2.7.10", minimum required is "2.7") 
-- Using C++03
-- Release build type: RelWithDebInfo
-- Adding UAVCAN STM32 platform driver
-- GIT_TAG = v1.5.0-565-g43a7f83
-- Configuring done
-- Generating done
-- Build files have been written to: /home/markw/git/kd0aij/PX4/Firmware/build_px4fmu-v3_default
PX4 CONFIG: /home/markw/git/kd0aij/PX4/Firmware/build_px4fmu-v3_default
ninja: Entering directory `/home/markw/git/kd0aij/PX4/Firmware/build_px4fmu-v3_default'
[26/529] Building CXX object CMakeFiles/msg_gen.dir/topics_sources/adc_report.cpp.obj
FAILED: /usr/lib/ccache/arm-none-eabi-g++   -DCONFIG_ARCH_BOARD_PX4FMU_V3 -D__DF_NUTTX -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -g -fno-exceptions -fno-rtti -std=gnu++0x -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Werror -Wextra -Wno-sign-compare -Wshadow -Wfloat-equal -Wpointer-arith -Wmissing-declarations -Wno-unused-parameter -Werror=format-security -Werror=array-bounds -Wfatal-errors -Werror=unused-variable -Werror=reorder -Werror=uninitialized -Werror=init-self -Werror=unused-but-set-variable -Wformat=1 -Wdouble-promotion -Werror=double-promotion -Wno-missing-field-initializers -fvisibility=hidden -include visibility.h -mcpu=cortex-m4 -mthumb -march=armv7e-m -mfpu=fpv4-sp-d16 -mfloat-abi=hard -nodefaultlibs -nostdlib -Os -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -O2 -g -DNDEBUG -I../src -I. -Isrc -I../src/modules -I../src/include -I../src/lib -I../src/platforms -I../src/drivers/boards/px4fmu-v3 -Isrc/modules/px4_messages -Isrc/modules -I../mavlink/include/mavlink -I../src/lib/DriverFramework/framework/include -I../src/lib/matrix -Ipx4fmu-v3/NuttX/nuttx-export/include -Ipx4fmu-v3/NuttX/nuttx-export/include/cxx -Ipx4fmu-v3/NuttX/nuttx-export/arch/chip -Ipx4fmu-v3/NuttX/nuttx-export/arch/common -Iexternal/Install/include    -Os -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -MMD -MT CMakeFiles/msg_gen.dir/topics_sources/adc_report.cpp.obj -MF CMakeFiles/msg_gen.dir/topics_sources/adc_report.cpp.obj.d -o CMakeFiles/msg_gen.dir/topics_sources/adc_report.cpp.obj -c topics_sources/adc_report.cpp
In file included from topics_sources/adc_report.cpp:37:0:
../src/platforms/px4_config.h:45:26: fatal error: board_config.h: No such file or directory
 #include <board_config.h>
                          ^
compilation terminated.
```

LorenzMeier, 2016-11-08:
Yes, you need to duplicate that. But not the whole directory structure. In addition: You can just include the v2 header in your v3 config.

kd0aij, 2016-11-08:
@LorenzMeier Thanks, looks like that's working OK. I'll clean up my branch and reopen #5821 soon.

LorenzMeier, 2016-11-13:
Done.

closed 2016-11-13 by LorenzMeier