PX4/PX4-Autopilot #4924: SITL segfault on OS X

AndreasAntener, 2016-06-27:
Currently getting a segfault when starting `gazebo_standard_vtol`, any ideas? Just `gazebo` works.

```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x2703 of process 30993]
0x00000001000bd5eb in simulator::Report<simulator::RawGPSData>::copyData(void*, int) ()
(gdb) backtrace
#0  0x00000001000bd5eb in simulator::Report<simulator::RawGPSData>::copyData(void*, int) ()
#1  0x0000000101d00390 in ?? ()
#2  0x0000000101d00510 in ?? ()
#3  0x0000000101d00590 in ?? ()
#4  0x00000001000143f4 in GPSSIM::receive (this=0x2b0, timeout=<optimized out>)
    at /Users/ada/UAV/src/Firmware/src/platforms/posix/drivers/gpssim/gpssim.cpp:279
#5  0x0000000101d00a20 in ?? ()
#6  0x0000000101d00390 in ?? ()
#7  0x0000000101d00510 in ?? ()
#8  0x0000000101d00590 in ?? ()
#9  0x0000000101d00390 in ?? ()
#10 0x0000000100014355 in GPSSIM::task_main (this=0x2b0)
    at /Users/ada/UAV/src/Firmware/src/platforms/posix/drivers/gpssim/gpssim.cpp:356
#11 0x0000000100019045 in entry_adapter (ptr=0x101d00630)
    at /Users/ada/UAV/src/Firmware/src/platforms/posix/px4_layer/px4_posix_tasks.cpp:104
#12 0x00007fff8932805a in ?? ()
#13 0x0000000000001003 in ?? ()
#14 0x0000000101e81000 in ?? ()
#15 0x0000000101e80f50 in ?? ()
#16 0x00007fff89327fd7 in ?? ()
#17 0x0000000000000000 in ?? ()
```

AndreasAntener, 2016-06-27:
Hm, also happens on Jenkins

dagar, 2016-06-27:
You could first try running it with address sanitizer before debugging. Sometimes it makes the problem really obvious.
export MEMORY_DEBUG=1

AndreasAntener, 2016-06-27:
Thanks, basically comes down to the same thing: https://gist.github.com/AndreasAntener/bc83f88c5ce2251b0740e0f0dcd3c487

AndreasAntener, 2016-06-27:
My bad, merge conflict in rcS, `simulator start -s` disappeared..

closed 2016-06-27 by AndreasAntener