PX4/PX4-Autopilot #4926: way to feed in vision attitude estimate in attitude_estimator_q or LPE?

tuofeichen, 2016-06-27:
Hey guys,
I'm wondering if there's a way to feed in a visual SLAM estimate of the attitude as a measurement (not just the heading, but potentially the whole rotation matrix), to `attitude_estimator_q` or ` LPE`?

`LPE`'s `vision.cpp` seems to only take care of translation, and `attitude_estimator_q` seems to only take the heading into account. 

Thank you. 

-FC

mhkabir, 2016-06-28:
You can modify the Q estimator for that.

jgoppert, 2016-07-01:
I'm gearing up to do this as well. I"ll take a look, shouldn't be too hard.

zzzzzq, 2016-12-22:
@mhkabir @jgoppert 
Hello!
I wonder can I feed vision estimation now? If still can't can you guys give me some advice how to modify Q estimator? 
Thank you very much!!!

jgoppert, 2016-12-23:
Now supported.

closed 2016-12-23 by jgoppert

tuofeichen, 2017-02-09:
Hi @jgoppert ,
I was wondering where is this supported? is it in the master or any specific branch? I assume its some modifications on the attitude-q estimator, but it (master) still looks like it only takes in external heading instead of the entire rotation from vision. 

Best 
FC

jgoppert, 2017-02-09:
@tuofeichen It just takes vision yaw, that is correct, I'm working on full vision attitude support with the IEKF estimator, which should be out soon.

tuofeichen, 2017-02-09:
got it, any other difference between IEKF and LPE?  I'm looking at the code now but would appreciate a high level motivation of why we switch

jgoppert, 2017-02-09:
IEKF is still in development and I wouldn't recommend  switching yet.

Instead of separate attitude (attitude_estimator_q) and position (LPE) kalman filters, it is combined (IEKF), so it has 21 states. It also has a rough dynamic model that is found through system identification of log data. So IEKF is much more complicated but should give better results when all of the bugs are eliminated.

tuofeichen, 2017-02-09:
Interesting. Got it thanks!