PX4/PX4-Autopilot #4929: Unable to use pos control when not using GPS

pickledgator, 2016-06-28:
A new commit has broken the ability to utilize position control nav state when gps_failure flag is true. This should probably be checking for valid_lpos or valid_gpos flags instead of only checking for gps_failure.

https://github.com/PX4/Firmware/blob/8a12dee125c37a3e48104877efd0f109d90b039b/src/modules/commander/state_machine_helper.cpp#L674

mhkabir, 2016-06-29:
Yup :) Would you mind sending in a PR?

dagar, 2016-06-29:
The fixed wing position controller needs global position for POSCTRL. You could use status->is_rotary_wing. I believe it properly changes with VTOL transitions.

julianoes, 2016-07-13:
@pickledgator can you please test #5058.

pickledgator, 2016-07-13:
Solves issue. Thanks for the PR, have been busy.

julianoes, 2016-07-15:
https://github.com/PX4/Firmware/pull/5058 is pushed.

closed 2016-07-15 by julianoes