PX4/PX4-Autopilot #4930: branch cleanup

dagar, 2016-06-28:
@LorenzMeier @bkueng @tumbili 

Hey guys, I don't want to distract from development, but one minor annoyance with working on Firmware is the overwhelming number of branches. There are 177 branches in the main repo, then doubled with a personal fork, plus personal branches, etc. It becomes quite difficult to find things. Can we try to clean it up a bit?

Here's a list of branches that can definitely be deleted.

     % git branch -a --merged master
    remotes/origin/auto_disarm
    remotes/origin/autopilot_capabilities
    remotes/origin/config_all_devices
    remotes/origin/control_state
    remotes/origin/ekf2_in_air
    remotes/origin/fw_landing_bugfix
    remotes/origin/getopt-cleanup-v2
    remotes/origin/i2c_100khz_init
    remotes/origin/logging_rate
    remotes/origin/mag_robustness
    remotes/origin/mavlink_act_ctrl
    remotes/origin/mavlink_recv
    remotes/origin/no_rc
    remotes/origin/nuttx_next_cmake_last
    remotes/origin/paramssrcparserpython3
    remotes/origin/ram_cleanup
    remotes/origin/rc_disabled
    remotes/origin/ros_mavlink
    remotes/origin/sitl_gazebo_gps
    remotes/origin/sitl_stable
    remotes/origin/sitl_startup_script
    remotes/origin/tracking
    remotes/origin/transition_flag_fix
    remotes/origin/two_point_curve
    remotes/origin/unknown_command
    remotes/origin/uorbtinymerge
    remotes/origin/vtol_cleanup
    remotes/origin/vtol_pwm_out

Here's a list of all the branches by age
https://gist.githubusercontent.com/dagar/261fad9c60f2357bb9ec1ffca795bf2a/raw/76aeae25ea473cb0b4685fe1aeac0af9d2055534/gistfile1.txt

Github stale branches is also helpful.
https://github.com/PX4/Firmware/branches/stale

bkueng, 2016-06-28:
Good point, I deleted my old branches.

LorenzMeier, 2016-06-28:
Thanks for the hint - I've also stripped a lot of branches, down to 94 now. @tumbili Would be good if you go through your very old branches as well.

LorenzMeier, 2016-06-29:
Down to 45. I'm closing this right now and will continue to narrow down PRs.

closed 2016-06-29 by LorenzMeier

dagar, 2016-06-29:
Much better. I can almost see them all on the same page.
Thanks!