PX4/PX4-Autopilot #4931: Quadplane: Altitude loss on pusher operation

LorenzMeier, 2016-06-28:
@sanderux Every time the pusher starts the system looses altitude. I believe it creates a static effect in the fuselage.

sanderux, 2016-06-29:
In my tests it was always increasing altitude. I think my cg was too far back and yours is too far foreard

LorenzMeier, 2016-06-29:
This happens while hovering 2 meters above the ground. I know for a fact that there is a static effect in the body. This has nothing to do with flight dynamics.

sanderux, 2016-06-29:
Doesnt the quad drive compensate for this?

LorenzMeier, 2016-06-29:
I already told you its a static pressure effect in the body. That means that the estimate goes bad. The only way to fix this is by either using a static port on the X4 body or adding ventilation holes which prevent the static effect. Issues with Physics have to be solved with physical solutions.

sanderux, 2016-06-29:
I will do some CFD testing before we start machine producing the modules to make sure no down force is created by them

LorenzMeier, 2016-06-29:
You can skip that. It happens at zero airspeed, so your modules are not involved. I don't think you understood the problem yet.

sanderux, 2016-06-29:
The pusher pulls air over the modules.
What do you think should change to address this?

LorenzMeier, 2016-06-29:
It changes the pressure inside the fuselage. I'll have a look with Roman into the log.

sanderux, 2016-06-29:
Ah, ofcourse. It is designed to pull air through the fuse to cool things. That will create low presure in the fuse when the pusher starts. The baro should in that case be isolated from this effect

tumbili, 2016-06-29:
Log: http://logs.uaventure.com/view/Zbk5bFUfgiBN5xtsoJNSCA

tubeme, 2016-06-29:
Instead can we just use the GPS Alt for this very moment of transition with higher wight in the math for the GPS alt instead of the baro alt. We have no classical fuselage like the X8 but rather a hood over the wing, and we experience similar loss of altitude during Altitude and Position transitions. It is not that bad though it is pretty small loss of altitude. After that it flies perfect.

sanderux, 2016-07-04:
@LorenzMeier it seems to me this would be a problem in most fixed-wings, it's only vtol where it becomes noticeable. to compensate for this you shield either use an external barometer or shield the entire FC. 

There is also a third option that seems a bit unconventional but i think it would work:
When the pusher enables there is usually no lateral movement. We could consider factoring in, or even switching to the static port from the pitot tube.

sanderux, 2016-07-23:
@LorenzMeier i examined the log @tumbili provided, your suggestion that pressure decreases when the pusher motor is activated does not seem to be backed by the log, the barometric readings match those of the gps. could there be an other reason?

[image]

tubeme, 2016-08-23:
We have flown ~70 mission with 1.3.4 so far. We can confirm the Altitude Loss in the transition. Most noticeable in Guided Modes and Mission mode. Less obvious in Manual.

Some logs:
http://logs.uaventure.com/view/zk3TpFujwzHyTD5dAjCF9h
[image]

http://logs.uaventure.com/view/447gnSEXrRyRe6YjHBoPef
[image]

http://logs.uaventure.com/view/CvQzw6v9bYSbFBDQtuPfq7
[image]

Probably temporary solution is to give some pitch in the transition with the params...

Any ideas where this comes from? It is for sure not the static pressure problem as @sanderux mentioned in the previous post.

sanderux, 2016-08-24:
@tubeme have you tried increasing blending speed?

tubeme, 2016-08-24:
So I have a dilemma looking at this. 
[image]

Now that you mention the blending speed I clearly see in the graphics that we loose the altitude the moment we reach the blending speed of 8m/s. At this point on the quad starts fade out, and during this period from 8m/s to the transition speed of 12 m/s. we have a gap for lift in the plane. The quad stops supporting the frame during this period and the plane just gor enough speed to generate lift. But this lift is not enough obviously. So during this very moment the plane should pitch up. But it does not. I clearly see the PiutchRateSP and the plane not following it. 

But where is my dilemma actually. As in previous posts we discussed the issue of blending the controls. We had the best results having blending all the time during quad mode even in hover mode (our first KK implementation was like this). And then stop the blending at the end of the transition. And actually the blending speed parameter states this:

_Airspeed at which we can start blending both fw and mc controls. Set to 0 to disable._

Which I understand as we blend after 8m/s. If I increase to let say 10 or even 12 m/s (the same as our transition speed) it will mean that we will not use the plane control surfaces up to the end of the transition. Which is not the case we want to achieve. 

Is there a way we could enable the blending all the time up to the end of the transition?

On the basis of what I see in the graphics you are correct we will most likely solve the altitude problem. But at the same time we will go further away from our goal to blend all the time in quad. That is my dilemma. :)

tubeme, 2016-08-24:
Also we have altitude increase in each and every back transition. Suppose this is due to pitchup command the quad is making to brake the speed. Because the plane is with a lot of speed and energy this transforms in altitude increase. That is not a problem but it is not a star wars kind of smooth transition to full stop :). 

What if we give a pitch up command with the quad and at the same time we give a pitch down command to the airplane? I think this will equal air-braking. What do you think?

sanderux, 2016-08-24:
Its worth inspecting but might lead to quad saturation. Im also considering telling the quad to increase altitude during transition, as this happens natively it will no longer go eratic to try and lower altitude but just accept it

tubeme, 2016-08-24:
For the front transition it is always better to increase the altitude than decrease it during transition. If it increases it automatic no problem. We could transition at 5 meters without worry. What we try to mimic is how the birds take off and land. This is the most effective way energy wise. 

We had a crash two weeks ago because of this. Just didn't notice that the take off is set to 20 meters in a new mission and we hit a tree.

tubeme, 2016-08-24:
You know the simple change for the front transition could be to trigger fade out of the quad thrust at transition airspeed not blending airspeed. At blending airspeed we start blending controls but we keep the quad full thrust until end of transition. This could solve the altitude loss problem.

tubeme, 2016-08-24:
For the back transition probably the best strategy for a smooth fast full stop would be at the flip of the transition button to cut the throttle, in a second or two to make a nasty pith up (flare) to stall the plane, and then full thrust the quad to level the machine. The pith up should be a pitch bang, not a full pitch that we hold. Just a bang to kill the speed. I do this pitch up bang during front transition sometimes and cancel the transition, it is that much energy it looses.  We could tune the timing with experiments and find which is the best sequence timing.

tubeme, 2016-08-30:
@sanderux We used 10 m/s blending in a decent amount of flights. No change! Looses altitude all the time. We made a brand new vtol to exclude mechanical problems. No change as well. I can post LOGS if you want. BTW the loosing altitude is so consistent that all the logs from one repetitive mission look almost the same. :)

tubeme, 2016-08-30:
@sanderux From the fligh LOG I gave you notice what is going on with the quad thrust even before the blending speed reached. It goead to almost 0.4. With such a thrust for sure it will loose altitude while not having lift up to 10 m/s.

[image]

AndreasAntener, 2016-08-30:
@tubeme I looked at the log you just shared. The altitude loss you are experiencing happens after the transition (we have the same issue). The reason is that your setup needs a pretty high pitch up in fixed-wing to fly level at the airspeed you were flying. TECS needs a few seconds until he reaches that the integrator.

At the end of your fixed-wing flight you fly level with, holding altitude nicely, with a pitch setpoint of 20 degrees (and airspeed around 11m/s).

Is your plane holding altitude in stabilized at that speed? If not, I suggest you increase your pitch setpoint offset. This is probably not going to help with this issue, but it would be good anyway.

We need to look into a smart way of initializing TECS right after the transition to solve this but it's tricky as the behaviour suddenly changes once the MC motors stop completely.

sanderux, 2016-08-30:
If your airframe requires this pitch offset you could also consider setting it as pitch trim in the params, or setting it by changing it hardware wise

dagar, 2016-10-17:
I still lose a few meters of altitude during forward transition. Part of it is the throttle jumping from transition (100%) to cruise (75%). 

TECS is only enabled once transition to FW finishes. @priseborough @tumbili Do you think we could do a better job initializing TECS to eliminate the loss of altitude?

[image]

[image]

priseborough, 2016-10-18:
@dagar I would start by initialising integrators for the specific energy and specific energy balance loops so that they match the desired handover values for throttle and pitch demand.

tumbili, 2016-10-18:
@dagar We can probably do a generic TECS init function which can be used for planes as well.
To my knowledge until now the integrators are zeroed.

dagar, 2017-04-12:
This can still be a minor issue, but it's generally much better than it was. Let's open a fresh issue if there's more to discuss.

closed 2017-04-12 by dagar