PX4/PX4-Autopilot #4940: px4flow testing, always reject position control

rawk-v, 2016-06-30:
Hello
I am testing px4flow in pixhawk. But always getting **reject position control**.

1.  Firmware version: **v1.3.3**, also tested in current 1.4.0rc2;
2.  px4flow attached to pixhawk with px4flow y direction and pixhawk forward remain the same;
3.  px4flow have been calibrated according to [https://pixhawk.org/modules/px4flow](https://pixhawk.org/modules/px4flow), I could always get px4flow quality above 230.

Really need some help, thanks!

jgoppert, 2016-07-01:
@rawk-v Have you tried setting  SYS_MC_EST_GROUP =1 and rebooting. A lot of the users that have been doing optical flow testing have been using this estimator and it should be more reliable.

jgoppert, 2016-07-01:
@rawk-v Also, set LPE_GPS_ON = false, or it will require you to have gps to initialize.

jgoppert, 2016-07-01:
@rawk-v, see my latest flight on https://github.com/PX4/Firmware/pull/4483, the param changes at the bottom are also important if you want to do auto missions with flow, also make sure you keep the altitude low (1-3) m so that the camera can get good flow readings.

dagar, 2016-07-01:
Should we do something to constrain altitude if using flow only in a position control mode?

jgoppert, 2016-07-01:
I lost flow at about five meters we can also unable flow based navigation without sonar.

jgoppert, 2016-07-01:
Maybe if estimated terrain distance is above sonar range.

mhkabir, 2016-07-02:
I think it also makes sense to introduce smart controller limits, especially as we move more and more towards navigating complex environments. 

Using estimation covariance to scale maximal trajectory following speed, for example makes a lot of sense. e.g for pure vision, the vehicle should slow down as the vision covariance goes up due to say lack of features in the environment. Or for flow only, limiting maximum altitude in the focus band of the lens. 

Even further work to take advantage of multiple perceptual sensor would be adding risk aware global planning, like someone is currently working on.

rawk-v, 2016-07-02:
@jgoppert Thank you! 
I have set SYS_MC_ESG_GROUP=1 before, not working. But after setting LPE_GPS_ON=0, it works like a charm. Thanks :).

rawk-v, 2016-07-06:
jgoppert, I am testing auto mission with optical flow now, I set all params as you suggest.
I takeoff the UAV manually and switch to position control successfully  at 1.2m approximately, but whenever I switch to automission channel, UAV just land and do nothing.

I have never test automission before, could you please show me some valid waypoint? Or any suggestions? Thanks!

LorenzMeier, 2016-12-28:
Closing as stale.

closed 2016-12-28 by LorenzMeier

saeidkazemi, 2017-02-12:
hi
We have some unknown problems to compile your code and we would be pleased if you could help us to fix it.
I have dealt with the problem of Position Control Mode actually the system rejects this mode.
We do not have GPS and want to run the Position Control with optical Flow Sensor.
We cannot find the LPE-GPS-on in Qground to turn it off, so how can we turn off the GPS?
We set the sys_group in LPE mode but again it rejects the position control mode. Do you know the reason?

Pixhawk version 2

LorenzMeier, 2017-02-12:
@saeidkazemi I'm going to block you from this issue tracker for excessive spamming. Filing an issue once is absolutely sufficient.

LorenzMeier, 2017-02-12:
Please open a thread on http://discuss.px4.io and provide enough feedback so the community can help you. Please do not file the same issue 10 times.

DroneMang, 2018-02-28:
Hi,

Trying to get the optical flow and lidar working to do position hold indoors without GPS. They appear to be working fine in the inspector window widget, I can see the distance changing for the rangefinder, and I can see values changing for the optical flow. When I switch to the position control flight mode, I get the "reject position control" flag. I have tried the below from previous posts, and set EKF2_AID_MASK to only use optical flow. I have tried setting SYS_MC_EST_GROUP to 1 and 2. The variable LPE_GPS_ON does not appear to exist. I am using a pixhawk with firmware 1.7.3 and QGroundControl daily build downloaded on 2/28/18 on a windows 10 machine. I've updated the optical flow camera to the firmware QGroundControl defaults to. I am talking to the LIDAR with PWM. Garmin says to use a 1k resistor, I'm using 470 ohm per the tutorial, and just want to point out that the order of the wires has reversed in the V3 device.

Any ideas? Thanks!

The pixhawk will not enter position control mode unless vehicle_local_position_ned messages (a.k.a. local_position_ned in mavlink) are regularly published by the estimator (LPE, in your case), with the flag xy_valid set.

You can disable the GPS in the LPE estimator by seting the parameter LPE_GPS_ON to zero. Also, check the value for LPE_FLW_QMIN, and make sure that this value is at or below the average quality measurement reported by the optical flow camera.

Also, you can try using EKF2 instead of LPE, if you do so make sure to change the EKF2_AID_MASK parameter to disable GPS and enable optical flow fusion.

Good luck.