PX4/PX4-Autopilot #4944: TAKEOFF command fly away if lat/long not NaN

vooon, 2016-06-30:
@TobiMiller found a problem with takeoff.

To reproduce run simulator and mavros:
```
1$ make posix_sitl_default jmavsim
2$ roslaunch mavros px4.launch fcu_url:=udp://@localhost:14557
3$ rosrun mavros mavsafety arm
3$ rosrun mavros mavcmd takeoffcur 0 0 20
```

And with QGC takeoff works, because it set coords to NAN. So `rosrun mavros mavcmd takeoff 0 0 nan nan 600` works.

I think problem is in navigator code.

vooon, 2016-06-30:
[navigator_main.cpp](https://github.com/PX4/Firmware/blob/master/src/modules/navigator/navigator_main.cpp#L468)
```
				rep->current.lat = cmd.param5 / (double)1e7;
				rep->current.lon = cmd.param6 / (double)1e7;
```

Is incorrect since param5/6 just copied in `mavlink_receiver.cpp`.

vooon, 2016-06-30:
`DO_REPOSITION` has same issue.

LorenzMeier, 2016-06-30:
Fixed in https://github.com/PX4/Firmware/pull/4947.

closed 2016-06-30 by LorenzMeier