PX4/PX4-Autopilot #4954: jMAVSim: mavlink_receiver.cpp:publish  hil_sensors combine but can not receive!

RizHuang, 2016-07-01:
in mavlink_receiver.cpp there is 
`	/* publish combined sensor topic */
		if (_sensors_pub == nullptr) {
			_sensors_pub = orb_advertise(ORB_ID(sensor_combined), &hil_sensors);

		} else {
			orb_publish(ORB_ID(sensor_combined), _sensors_pub, &hil_sensors);
		}`
But can not receive from position_estimator.cpp

LorenzMeier, 2016-07-01:
Have you checked that you are receiving HIL sensors?

RizHuang, 2016-07-01:
yes ! I am sure I have receive HIL sensors !

RizHuang, 2016-07-01:
I tested with the Firmware master branch code  and jMASim master code .

RizHuang, 2016-07-01:
I find something wrong here in the position_estimator_inav.cpp . 
before the main task , there is a poll use to waiting 3s for baro data to caculate baro offset . if i comment these code  and direct give the baro_offset , it work well

dagar, 2017-04-11:
Is this working now?

dagar, 2017-04-18:
Hi @RizHuang can you try this again and reopen if it's still an issue?

closed 2017-04-18 by dagar