PX4/PX4-Autopilot #4955: New feature requested, single motor run/check, to find out if m1 -m8 are connected properly

JPluess, 2016-07-01:
Hi dear developers,

i have just set up a drone with the tarot zyx-m controller, i see they have a nice tool that can spin each motor, this way it is easy to identify if al motors are connected properly.
sometimes wiring is hidden, this would be of much help.

[image]

press m1 -> m1 will spin slowly for 5 sec.

thanks a lot

julianoes, 2016-07-07:
There is one command line tool to do just this, however it's not as easy as the thing on the screenshot.

Remove the SD card from the Pixhawk and connect over USB. Open a serial console and you should get to the `nsh>`. At this point, press the safety switch, so it double-blinks.

Then type:
```
pwm arm
```

And to test one channel, do:
```
pwm test -c 1 -p 1300
```
where 1300 is your PWM value and 1 is your channel counting from 1.
Also see the docs here: https://pixhawk.org/firmware/apps/pwm

dagar, 2016-07-07:
I opened a pull request to move MAV_CMD_DO_MOTOR_TEST into common. Don was interested in adding this QGC.

https://github.com/mavlink/mavlink/pull/580

dagar, 2016-07-07:
Would it make sense to modify motor_ramp to do this? Plus something to temporarily disable the controllers.

dagar, 2016-07-10:
https://github.com/PX4/Firmware/issues/5023

LorenzMeier, 2016-07-11:
I'm closing the feature request in favour of the issue discussing how we will implement it.

dagar, 2019-01-21:
Duplicate of https://github.com/PX4/Firmware/issues/10782

closed 2019-01-21 by dagar