PX4/PX4-Autopilot #4956: Dronekit mission test script needs to use test framework to produce test result artifacts

AndreasAntener, 2016-07-01:
e.g. Pythons `unittest`
Then we can integrate it in CI

LorenzMeier, 2016-07-01:
Nice! Volunteers? :)

dagar, 2016-07-01:
Has anyone talked to Ardupilot Dronekit users? It seems like testing at the Dronekit level could be shared between projects.

MikaelFerland, 2016-07-25:
Maybe I'm late a bite, but last time I tried Dronekit, it did not work pretty well. So, I've investigated to make it works. To get the initial connection I have modified this line https://github.com/dronekit/dronekit-python/blob/master/dronekit/__init__.py#L1093. The modification added the boolean `and (self._autopilot_type != mavutil.mavlink.MAV_AUTOPILOT_PX4` in the if statemetn so we were able to keep the HEARTBEART. After all that, I found a roadblock that was the mode handling with the custom_mode field to be able the get the right mode. An finally, I found also that we have the modify the setter for the mode because they are not considering the mode for the PX4. In Dronekit, they choose the right list considering the platform assigned into the FC. This just a little report that I've found maybe it could help you if you want to add a unit test.

dagar, 2016-07-26:
@MikaelFerland I think modes are the main piece missing right now. Would you be interested in adding PX4 modes and getting the change into dronekit? I can help you get the modes straight.
https://github.com/PX4/Firmware/issues/4224

MikaelFerland, 2016-07-27:
@dagar I could help for sure, but I could look at it just after my two weeks rush of summer school. They have a talk about the connection of dronekit here https://github.com/dronekit/dronekit-python/issues/632#issuecomment-228872603. I have a part of the solution, but it's a specific case to make a workaround to isolate the problem with the mode. Next time, I will look into the MAVros project to see how they manage the mode. After that to make it clean, I think we will need to patch pymavlink.

MikaelFerland, 2016-08-14:
@dagar after our little talk last time, @squilter did the px4 modes work in pymavlink. It helped me a lots. So my exams are finished and the modes are working into dronekit. I did something here https://github.com/dronekit/dronekit-python/pull/652. I will check soon why the build have failed.

closed 2018-01-17 by dagar