PX4/PX4-Autopilot #4957: Clean up flash space on FMUv2

LorenzMeier, 2016-07-01:


LorenzMeier, 2016-07-01:
134966500 00000256 t crc8_tab
134978900 00000256 t crc_table

LorenzMeier, 2016-07-01:
135017752 00000256 t mavlink_sha256_constant_256
135020612 00000256 t mavlink_sha256_constant_256
135023928 00000256 t mavlink_sha256_constant_256
135026028 00000256 t mavlink_sha256_constant_256
135027692 00000256 t mavlink_sha256_constant_256
135028148 00000256 t mavlink_sha256_constant_256
135028640 00000256 t mavlink_sha256_constant_256

davids5, 2016-07-05:
Colaless Mavlink: Due to static inline mavlink/include/mavlink/v2.0/checksum.h

coalesce saves 224

08024ed4 0000001c t crc_accumulate
08038d5e 0000001c t crc_accumulate
0803b530 0000001c t crc_accumulate
0803cb58 0000001c t crc_accumulate
0803d67e 0000001c t crc_accumulate
0804166c 0000001c t crc_accumulate
08044e20 0000001c t crc_accumulate
08045c4c 0000001c t crc_accumulate

coalesce saves 1792
080cbef4 00000100 t mavlink_sha256_constant_256
080cca24 00000100 t mavlink_sha256_constant_256
080cd6f8 00000100 t mavlink_sha256_constant_256
080cdf54 00000100 t mavlink_sha256_constant_256
080ce7c8 00000100 t mavlink_sha256_constant_256
080ce994 00000100 t mavlink_sha256_constant_256
080ceb80 00000100 t mavlink_sha256_constant_256

080c0ba4 00000100 t crc8_tab protocol.h
080c3df4 00000100 t crc_table trone.cpp

coalesce Save 2520:
08038e5c 000001a4 t mavlink_sha256_update
0803b5a8 000001a4 t mavlink_sha256_update
0803cbc4 000001a4 t mavlink_sha256_update
0803dc60 000001a4 t mavlink_sha256_update
08041688 000001a4 t mavlink_sha256_update
08044e80 000001a4 t mavlink_sha256_update
08045c8c 000001a4 t mavlink_sha256_update

Investigate:
0802847c 00000050 t __px4_log_modulename.constprop.0
08077074 00000050 t __px4_log_modulename.constprop.0
08025bc8 00000050 t __px4_log_modulename.constprop.4
08078148 00000050 t __px4_log_modulename.constprop.4
0802d658 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.0]
0805db98 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.0]
080717c4 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.0]
08071f00 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.0]
0807dd08 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.0]
08089290 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.0]
0800421c 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.1]
08018c88 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.1]
0801a7f8 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.1]
08062214 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.1]
08063280 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.1]
08063888 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.1]
08064604 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.1]
08072cec 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.1]
0808e23c 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.1]
080940f4 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.1]
08041a9c 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.11]
0807ae00 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.11]
0806b9bc 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.12]
0806a1fc 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.13]
08033944 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.14]
080222b0 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.2]
0802dfe8 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.2]
080604c4 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.2]
08061af8 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.2]
0807a914 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.2]
0807cca4 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.2]
0807fec8 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.2]
080bcec8 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.2]
080709cc 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.23]
0801bc68 00000050 t __px4_log_modulename(int, char const*, ...) [clone .constprop.3]
[... 58 lines cut ...]
080296cc 00000054 t __px4_log_modulename(int, char const*, ...)
0802e47c 00000054 t __px4_log_modulename(int, char const*, ...)
0802f51c 00000054 t __px4_log_modulename(int, char const*, ...)
0802fbbc 00000054 t __px4_log_modulename(int, char const*, ...)
08030b78 00000054 t __px4_log_modulename(int, char const*, ...)
080310b0 00000054 t __px4_log_modulename(int, char const*, ...)
08031ba4 00000054 t __px4_log_modulename(int, char const*, ...)
08032650 00000054 t __px4_log_modulename(int, char const*, ...)
08038d7c 00000054 t __px4_log_modulename(int, char const*, ...)
080469a8 00000054 t __px4_log_modulename(int, char const*, ...)
08055830 00000054 t __px4_log_modulename(int, char const*, ...)
08059870 00000054 t __px4_log_modulename(int, char const*, ...)
0805b988 00000054 t __px4_log_modulename(int, char const*, ...)
080612e4 00000054 t __px4_log_modulename(int, char const*, ...)
0807a00c 00000054 t __px4_log_modulename(int, char const*, ...)
0808c0dc 00000054 t __px4_log_modulename(int, char const*, ...)
0801061c 00000054 t __px4_log_modulename(int, char const*, ...) [clone .constprop.0]
0807c580 00000054 t __px4_log_modulename(int, char const*, ...) [clone .constprop.0]
0802456c 00000054 t __px4_log_modulename(int, char const*, ...) [clone .constprop.1]
08024790 00000054 t __px4_log_modulename(int, char const*, ...) [clone .constprop.1]
08093f0c 00000054 t __px4_log_modulename(int, char const*, ...) [clone .constprop.1]
0807e518 00000054 t __px4_log_modulename(int, char const*, ...) [clone .constprop.18]
0801ac90 00000054 t __px4_log_modulename(int, char const*, ...) [clone .constprop.2]
0802394c 00000054 t __px4_log_modulename(int, char const*, ...) [clone .constprop.3]
0800ed6c 00000054 t __px4_log_modulename(int, char const*, ...) [clone .constprop.5]

davids5, 2016-07-05:
@LorenzMeier 

what commands would we never use:
 [
 dirname
?
basename
break
cat
cd
cmp
cp
date
dd
df
echo
exec
exit
free
help
hexdump
kill
ls
mb
mh
mkdir
mkfatfs
mkfifo
mkrd
mount
mv
mw
ps
pwd
rm
rmdir
set
sh
sleep
test
time
umount
uname
unset
usleep
xd
true
false

Please put an X in front of once I can remove from NuttX and I will disable them in the configs

davids5, 2016-07-06:
@LorenzMeier 

Reduced command set to 

?
[
break
cat
cd
cmp
cp
date
df
echo
exec
exit
free
help
kill
ls
mkdir
mkfatfs
mount
mv
mw
ps
pwd
rm
rmdir
set
sh
sleep
test
time
umount
unset
usleep
false
true

Saves 4344 Bytes

davids5, 2016-07-06:
@LorenzMeier  this https://github.com/PX4/Firmware/pull/4994 freesup 4506 Bytes

closed 2016-11-19 by LorenzMeier