PX4/PX4-Autopilot #4962: px4flow testing, position control failed under I-NAV in v1.3.3 and above

dengzhijie, 2016-07-02:
Hello
I am testing px4flow (no GPS) on pixhawk which running I-NAV.It can be switched to posctl successfully(not ARMED).But after it was armed,when i changed to position control, the light turned to green and then to pink(purple?),and failed to hover.
Firmware version: v1.3.3, v1.4.0rc2,also tested in latest 1.4.0rc3, all failed.
 expecting for you help,thanks!

As a comparason, iNAV works without any problem using firmware v1.3.0/v1.3.1 or earlier. So anyting changed?
@iZhangHui @rolandash @eflyinghi

dengzhijie, 2016-07-04:
We tested v1.3.4 and v1.4.1 today,but the problem still remain.

LorenzMeier, 2016-07-04:
I quickly looked at the diff between 1.3.1 and 1.3.4 and can't really see anything substantial. Can you please re-test with 1.3.1 and make sure it still works?

dengzhijie, 2016-07-05:
@LorenzMeier I am sorry,we made a mistake.There are no problem of v1.3.1,v1.3.3 and v1.3.4.So the problem only exisits in v1.4.0 and the latest v1.4.1.

dengzhijie, 2016-07-05:
I changed the title just now to reflect the latest testing result.

LorenzMeier, 2016-07-05:
That makes a lot more sense. Can you try with the LPE estimator, because we will stop developing INAV soon?

dengzhijie, 2016-07-05:
@LorenzMeier We tested LPE under v1.4.0 and the latest v1.4.1,the same phenomenon to I-NAV

LorenzMeier, 2016-07-05:
Can you please read this page and check if you followed all recommendations?
http://dev.px4.io/flow_lidar_setup.html

iZhangHui, 2016-07-06:
@LorenzMeier This issue is caused by  [code here.](https://github.com/PX4/Firmware/blob/master/src/modules/commander/state_machine_helper.cpp#L674)
It only checks for gps_failure, so POSCTL flying with px4flow (no GPS) will always fall into fail-safe mode.

I'm able to fly when set parameter CBRK_GPSFAIL to 240024.

LorenzMeier, 2016-07-06:
@julianoes can you look into this?

julianoes, 2016-07-06:
Aha, I didn't think of this case. Is it correct that POSCTL should be allowed with a local position estimate whereas the auto modes should not be allowed without GPS?

LorenzMeier, 2016-07-06:
Yes, pretty much

julianoes, 2016-07-13:
@dengzhijie can you please #5058.

dengzhijie, 2016-07-15:
@julianoes We  tested LPE under the v1.4.1rc2 on pixhawk and MindPX  today,the problem was solved and it successed to POSCTL .
Really thank to your help!

julianoes, 2016-07-15:
@dengzhijie thanks for testing and reporting. I'll merge it then.

julianoes, 2016-07-15:
Resolved with https://github.com/PX4/Firmware/pull/5058.

closed 2016-07-15 by julianoes