PX4/PX4-Autopilot #4963: px4flow testing,pixhawk running LPE,always reject altitude control under 1.4.0rc2 and 1.4.0 rc3

dengzhijie, 2016-07-02:
Hello
I am testing px4flow(no GPS) on pixhawk which running LPE.But always reject altitude control.
 Firmware version: v1.4.0rc2, also tested in latest v1.4.0rc3;
In addition,position control also failed under LPE under v1.3.3 and above.
 expecting for you help,thanks!

As comparason, LPE can work without problem under both altctl and posctl in v1.3.1. same device and same testing environment.
@iZhangHui @rolandash@eflyinghi

rolandash, 2016-07-02:
should @jgoppert

jgoppert, 2016-07-02:
@dengzhijie Please try 1.4.0 rc4, that has my latest updates to LPE. Also I just spent a lot of time improving the documentation for flow with LPE, so please visit the webpage here and see if it helps you: http://dev.px4.io/flow_lidar_setup.html

dengzhijie, 2016-07-04:
@jgoppert We tested v1.4.1 today(there was no firmware package of v1.4.0 rc4),but the problem seems was not solved,We followed your advice to view the document and setting SYS_MC_EST_GROUP =1 and LPE_GPS_ON = false(no GPS connected, only flow attached),but it did not work.

I also tested the case when GPS and flow are all disconnected. ALTCTL still rejected.

The only success is when GPS connected, then ALTCTL / POSCLT both work.

Is the same situation on your side? any advices
Thanks.

@rolandash @iZhangHui @eflyinghi

dengzhijie, 2016-07-04:
Forgetting to say that  ALTCTL/POSCTL  both works under v1.3..4 when using flow only and no GPS.

jgoppert, 2016-07-04:
Can you attach a log?

dengzhijie, 2016-07-05:
@jgoppert We've solved the problem of  ALTCTL rejecting today ，and it was caused of the version of qgc which not saved the param of  LPE_GPS_ON rightly,and when I changed another version of qgc,the param was saved collectly.My qgc sierial number is v2.9.7b-369-g25a4502.
Really thank to your help!

closed 2016-07-06 by dengzhijie

saeidkazemi, 2017-02-12:
hi
We have some unknown problems to compile your code and we would be pleased if you could help us to fix it.
I have dealt with the problem of Position Control Mode actually the system rejects this mode.
We do not have GPS and want to run the Position Control with optical Flow Sensor.
We cannot find the LPE-GPS-on in Qground to turn it off, so how can we turn off the GPS?
We set the sys_group in LPE mode but again it rejects the position control mode. Do you know the reason?

Pixhawk version 2