PX4/PX4-Autopilot #4968: Latest master (d96810f) has compilation errors for px4fmu-v2_default

vvranjek, 2016-07-03:
It appears that the following types are missing in the `vehicle_gps_position.msg ` 
```
uint64 timestamp_position
uint64 timestamp_time
uint64 timestamp_variance
uint64 timestamp_velocity
```

After I added those I went on to compile and there were quite a few problems in ekf2. Some missing/undefined types such as `ext_vision_message` , `ev_delay_ms` and others. I couldn't go on to "fix" the problem as there were errors I couldn't have resolved. I did the `git submodule update --init --recursive` after merge/checkout.

Could there be anything else I am missing or maybe you forgot to commit submodule updates or something similar?

mhkabir, 2016-07-03:
Master being broken is practically impossible, thanks to continuous integration. 

Are you on an absolutely clean pull of master? Run  ```make clean```, ```make submodulesclean``` and ```git clean - dfx``` to make sure.

vvranjek, 2016-07-03:
`make clean` and `make submodulesclean` did it for me, thanks! Case closed.

closed 2016-07-03 by mhkabir