PX4/PX4-Autopilot #4970: Submodules not updating

vvranjek, 2016-07-03:
Hi,

Not sure if this is a PX4 problem or a git one, but lately I can't get to update the submodules on my fork.

Here is what I do:
```
make clean
checkout upstream/master
make submodulesclean
git submodule update --init --recursive
make
```
Everything is good. But when I go to my master, I do:
```
git checkout master
git fetch upstream
git merge upstream/master
make clean
make submodulesclean
git submodule update --init --recursive
make
```
It gives me compilation errors. I tried debugging the issue came to the conclusion that it's a submodule problem. Here are the outputs of `make submodulesclean` for `upstream/master` and my forked `master`. You can see that the submodule commits being checked out aren't the same.

upstream/master:
```
Submodule path 'NuttX': checked out 'f0f4bdc872d324f64e9d93d6f8989d3c1dfa2633'
Submodule path 'Tools/gencpp': checked out '26a86f04bcec0018af6652b3ddd3f680e6e3fa2a'
Submodule path 'Tools/genmsg': checked out '72f0383f0e6a489214c51802ae12d6e271b1e454'
Submodule path 'Tools/jMAVSim': checked out 'a5b0f1f0896c9372c62222e88117c6ced2c6282b'
Submodule path 'Tools/jMAVSim/jMAVlib': checked out '53f51ed9969472c82119ff2edde9c62b7b70eb94'
Submodule path 'Tools/sitl_gazebo': checked out '82b354875040d97a3f598f0dc260ddc9ca4a95b3'
Submodule path 'cmake/cmake_hexagon': checked out 'eb1d242e57b34a28ba4b8ab4c04901bd85540d07'
Submodule path 'mavlink/include/mavlink/v1.0': checked out '6adda4dbc1b68a29b8729c6e7ab34c9a1851c4df'
Submodule path 'mavlink/include/mavlink/v2.0': checked out '1e2920c68df46c25b7a14ca6cc47a478c6081d81'
Submodule path 'src/drivers/gps/devices': checked out 'a5e3e263aaaa613e007717d3ebf5ca7c597bcf51'
Submodule path 'src/lib/DriverFramework': checked out 'd1ab15894a147ecd9fce021f17dd591b352e126c'
Submodule path 'src/lib/DriverFramework/cmake/cmake_hexagon': checked out 'c0ba14e5983fd197cfff8baa70224d1d67908f0c'
Submodule path 'src/lib/DriverFramework/dspal': checked out 'b429a142bf6a21747d22b375d765cbf816add318'
Submodule path 'src/lib/DriverFramework/dspal/test/cmake_hexagon': checked out 'c0ba14e5983fd197cfff8baa70224d1d67908f0c'
Submodule path 'src/lib/ecl': checked out '5e0a229ee8bf09430051090f58f71c95fa4755f0'
Submodule path 'src/lib/ecl/matrix': checked out '95e3d7d6ce5274d9b7fec76c0093dc5eaa381abf'
Submodule path 'src/lib/matrix': checked out '3c87ae78ff4d80d1643b72afd2783478e87b8e5b'
Submodule path 'src/modules/uavcan/libuavcan': checked out 'b04396ace50155573e545ed9bf2fb09964ee2367'
Submodule path 'src/modules/uavcan/libuavcan/dsdl': checked out '9804a3e6972825586be252ce08dd899f44994b14'
Submodule path 'src/modules/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan': checked out 'c58477a644d20ccf95a20c151f3a0402f271c3b8'
Submodule path 'src/modules/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan/dsdl': checked out 'fd12483ddd4e58242d61d74a163e7aeaa1e0f466'
Submodule path 'unittests/googletest': checked out 'c99458533a9b4c743ed51537e25989ea55944908'

```

master:
```
Submodule path 'NuttX': checked out 'f0f4bdc872d324f64e9d93d6f8989d3c1dfa2633'
Submodule path 'Tools/gencpp': checked out '26a86f04bcec0018af6652b3ddd3f680e6e3fa2a'
Submodule path 'Tools/genmsg': checked out '72f0383f0e6a489214c51802ae12d6e271b1e454'
Submodule path 'Tools/jMAVSim': checked out 'a5b0f1f0896c9372c62222e88117c6ced2c6282b'
Submodule path 'Tools/jMAVSim/jMAVlib': checked out '53f51ed9969472c82119ff2edde9c62b7b70eb94'
Submodule path 'Tools/sitl_gazebo': checked out '9353834b2f89d08beaa055fc97321e1871bc2b3a'
Submodule path 'cmake/cmake_hexagon': checked out '77d487dc076a8519f81a335a5e4a2040799790de'
Submodule path 'mavlink/include/mavlink/v1.0': checked out '8106ab6a5853926073f399a5534374ecd8dd14c7'
Submodule path 'mavlink/include/mavlink/v2.0': checked out '699a2b0df1b259a962a01af788e8306ee8d016d8'
Submodule path 'src/drivers/gps/devices': checked out '8f2d911a5203e7f6d29c1e3cf3405972a87215d8'
Submodule path 'src/lib/DriverFramework': checked out '62449aef6decd2faa75bc2abd509333df5a675c4'
Submodule path 'src/lib/DriverFramework/cmake/cmake_hexagon': checked out 'da4d6e7898880ce370dca21dac1362cd56274a79'
Submodule path 'src/lib/DriverFramework/dspal': checked out '8f39699c6b79447f32f0714629da790bbf271aa5'
Submodule path 'src/lib/DriverFramework/dspal/test/dspal_tester/cmake_hexagon': checked out 'ba681fcd0d4c5ac95636dd8e9d8536dbd81beabd'
Submodule path 'src/lib/ecl': checked out '172f4be594a0b33997a07f5b46899c141a6cee86'
Submodule path 'src/lib/ecl/matrix': checked out '95e3d7d6ce5274d9b7fec76c0093dc5eaa381abf'
Submodule path 'src/lib/matrix': checked out '07fba8322a16cb2dac47e6a8ac7d21ec346313ff'
Submodule path 'src/modules/uavcan/libuavcan': checked out 'b04396ace50155573e545ed9bf2fb09964ee2367'
Submodule path 'src/modules/uavcan/libuavcan/dsdl': checked out '9804a3e6972825586be252ce08dd899f44994b14'
Submodule path 'src/modules/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan': checked out 'c58477a644d20ccf95a20c151f3a0402f271c3b8'
Submodule path 'src/modules/uavcan/libuavcan/libuavcan/dsdl_compiler/pyuavcan/dsdl': checked out 'fd12483ddd4e58242d61d74a163e7aeaa1e0f466'
Submodule path 'unittests/googletest': checked out 'c99458533a9b4c743ed51537e25989ea55944908'
```

I should probably note that I've been using this procedure for a few months now and it always worked fine using only `git submodule update --init --recursive` after merge. I have no idea what to do here...

dagar, 2016-07-03:
Updating submodules just checks out the version stored in your Firmware branch. I'm guessing at some point during a merge or something they didn't get updated and were accidentally committed. 

You'll have to get the them back to the state of master. Depending on what you're doing locally it might be easier to start a new branch from upstream/master and just pull over your local changes. Otherwise you can go through each different submodule and 

    git checkout upstream/master mavlink/include/mavlink/v1.0

You can do `make distclean` which cleans everything, syncs and updates submodules in one command.

LorenzMeier, 2016-07-03:
Also make sure to run ```git submodule sync```.

closed 2016-07-03 by LorenzMeier

vvranjek, 2016-07-03:
Ok, so `make distclean` didn't help but `git checkout upstream/master ...` did. After that I could merge the `master' to my other branch and it updated the submodules correspondingly. Tried (on another branch) `git submodule sync` and 'git submodule update --init --recursive` and it checked out the same old modules, so that didn't help I guess. I have no idea how that get messed up as I only use the `master` branch to sync with `upstream/master` and nothing else. Never touched the submodules. I guess git submodules are still a mystery to me. Any ideas how to prevent this in the future? Other then that, thanks for the help!

dagar, 2016-07-03:
make distclean = clean + submodules sync + submodules update

In git each submodule (eg src/lib/ecl) is just a file that points to a git commit. In your branch that pointer has changed. You need to revert that change or force each submodule to match the commit used in master.

For each submodule that needs to be fixed

    git checkout -f upstream/master mavlink/include/mavlink/v1.0
    git checkout -f upstream/master mavlink/include/mavlink/v2.0
    etc

Then commit the change and run `make distclean` to make sure everything is checked out cleanly to build. Submodules can be confusing, but the alternatives are worse.

vvranjek, 2016-07-03:
Yes, but shouldn't at least 'git status' tell me that a submodule is lagging behind? Instead I lost many hours figuring out why the compilation had errors...

dagar, 2016-07-03:
Yes git status would show you if it had just changed, but my guess is that you changed it accidentally in a previous commit.

Do you have a diff tool that can show you the diff of the entire tree?
Some git clients can do this (like eclipse) or you can use meld and run this.

    git difftool --tool=meld --dir-diff upstream/master

This shows everything in your local branch that's different than upstream. You'll see the submodules are set to something different and need to be fixed. Change them to match upstream/master, commit the change, make distclean, build.