PX4/PX4-Autopilot #4975: jmavsim

yota-code, 2016-07-04:
Newbie here,
I'm doing my best to follow [this page](http://dev.px4.io/simulation-sitl.html) to setup a basic SITL

`make posix_sitl_default jmavsim`

when I launch this I got:

```
PX4 CONFIG: posix_sitl_default
[1/1] cd /home/yota/disk/forge/Firmware && Tools/sitl_run.sh posix-configs/SITL/init/rcS none jmavsim none /home/yota/disk/forge/Firmware/build_posix_sitl_default
SITL ARGS
rc_script: posix-configs/SITL/init/rcS
debugger: none
program: jmavsim
model: none
build_path: /home/yota/disk/forge/Firmware/build_posix_sitl_default
empty model, setting iris as default
Buildfile: /media/yota/af65daa0-9a8d-4096-8909-e10f0c568391/forge/Firmware/Tools/jMAVSim/build.xml

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 0 seconds
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

Error occurred during Canvas3D callback:
java.lang.BootstrapMethodError: java.lang.NoClassDefFoundError: java/lang/invoke/StringConcatFactory
	at me.drton.jmavsim.Visualizer3D$CustomCanvas3D.postRender(Visualizer3D.java:883)
	at javax.media.j3d.Renderer.doWork(Renderer.java:1376)
	at javax.media.j3d.J3dThread.run(J3dThread.java:271)
Caused by: java.lang.NoClassDefFoundError: java/lang/invoke/StringConcatFactory
	... 3 more
Caused by: java.lang.ClassNotFoundException: java.lang.invoke.StringConcatFactory
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 3 more
```
From my modest java knowledge I guess it have to do with some classes not available in a mysterious $PATH_SOMETHING. It may even be linked to the options found [here](https://pixhawk.org/dev/hil/jmavsim) : `java -Djava.ext.dirs= -cp lib/*:out/production/jmavsim.jar me.drton.jmavsim.Simulator -udp 127.0.0.1:14560`.

But I can't go further, my jmavsim simulation fails and the make commands is a dead ends.
Thank you for you support

dagar, 2016-07-04:
What version of java do you have? java -version

yota-code, 2016-07-05:
```
$ java -version
openjdk version "1.8.0_91"
OpenJDK Runtime Environment (build 1.8.0_91-8u91-b14-0ubuntu4~16.04.1-b14)
OpenJDK Server VM (build 25.91-b14, mixed mode)
```
Here are the installed JVM... 
```
There are 2 choices for the alternative java (providing /usr/bin/java).

  Selection    Path                                           Priority   Status
------------------------------------------------------------
  0            /usr/lib/jvm/java-9-openjdk-i386/bin/java       1091      auto mode
* 1            /usr/lib/jvm/java-8-openjdk-i386/jre/bin/java   1081      manual mode
  2            /usr/lib/jvm/java-9-openjdk-i386/bin/java       1091      manual mode
```

In fact, with java-9-openjdk-i386 I get even nastier errors :)
```
Bug 1004: Caught: class com.jogamp.nativewindow.awt.AppContextInfo cannot access class sun.awt.AppContext (in module java.desktop) because module java.desktop does not export sun.awt to unnamed module @1763516
java.lang.IllegalAccessException: class com.jogamp.nativewindow.awt.AppContextInfo cannot access class sun.awt.AppContext (in module java.desktop) because module java.desktop does not export sun.awt to unnamed module @1763516
```

I cannot find out where should be StringConcatFactory defined and if something is missing in my java install o_O

dagar, 2016-07-05:
I'm not sure what to suggest. It looks like we have the exact same version of Ubuntu and java 8. The only obvious difference I see so far is that I only have java 8.

     % dpkg -l | grep openjdk
    ii  openjdk-8-jdk:amd64                        8u91-b14-0ubuntu4~16.04.1                                   amd64        OpenJDK Development Kit (JDK)
    ii  openjdk-8-jdk-headless:amd64               8u91-b14-0ubuntu4~16.04.1                                   amd64        OpenJDK Development Kit (JDK) (headless)
    ii  openjdk-8-jre:amd64                        8u91-b14-0ubuntu4~16.04.1                                   amd64        OpenJDK Java runtime, using Hotspot JIT
    ii  openjdk-8-jre-headless:amd64               8u91-b14-0ubuntu4~16.04.1                                   amd64        OpenJDK Java runtime, using Hotspot JIT (headless)

yota-code, 2016-07-12:
I suggest to close it, thank you for your support.

I started from scratch on a much cleaner ubuntu install (16.04 64 bit and openjdk-8-jre-headless openjdk-8-jdk-headless packages) and everything worked fine...

closed 2016-07-12 by yota-code

dagar, 2016-07-12:
I'm glad you were able to get it working.

Neels001, 2017-05-04:
Hello yota-code I am facing the same issue.Could you please help me with steps how to fix it.
Thank you!!

yota-code, 2017-05-04:
If I remember well, the problem disappeared when I had the correct java version, the ones mentioned on July the 12th: openjdk-8-jre-headless openjdk-8-jdk-headless.
No other ubuntu package gave me a good result... Do you use ubuntu too ?

Neels001, 2017-05-04:
Thank you for ur prompt reply..
Yes I use Ubuntu too, I installed same packages..still I face the same error..Do I need to uninstall java and then reinstall it?

yota-code, 2017-05-04:
you can try to find all already installed java packages and do an apt purge <package> to do a clean reinstall, and you could also provide : 
`java -version` and `lsb_release -a`
because I have no new idea :)

Neels001, 2017-05-04:
hi yota-code
I have provided the details of my problem.
The below are versions I am using..
neelesh@neelesh-Lenovo-G580:~$  lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.2 LTS
Release:	16.04
Codename:	xenial
neelesh@neelesh-Lenovo-G580:~$ 
neelesh@neelesh-Lenovo-G580:~$ java -version
openjdk version "1.8.0_121"
OpenJDK Runtime Environment (build 1.8.0_121-8u121-b13-0ubuntu1.16.04.2-b13)
OpenJDK 64-Bit Server VM (build 25.121-b13, mixed mode)
neelesh@neelesh-Lenovo-G580:~$ 
Errors I am getting:
copy_res:

BUILD SUCCESSFUL
Total time: 0 seconds
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

libEGL warning: bad surface attribute 0x3080
libEGL warning: bad surface attribute 0x3080
libEGL warning: bad surface attribute 0x3080
libEGL warning: bad surface attribute 0x3080
Error occurred during Canvas3D callback:
java.lang.BootstrapMethodError: java.lang.NoClassDefFoundError: java/lang/invoke/StringConcatFactory
	at me.drton.jmavsim.Visualizer3D$CustomCanvas3D.postRender(Visualizer3D.java:883)
	at javax.media.j3d.Renderer.doWork(Renderer.java:1376)
	at javax.media.j3d.J3dThread.run(J3dThread.java:271)
Caused by: java.lang.NoClassDefFoundError: java/lang/invoke/StringConcatFactory
	... 3 more
Caused by: java.lang.ClassNotFoundException: java.lang.invoke.StringConcatFactory
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 3 more
Error occurred during Canvas3D callback:
java.lang.BootstrapMethodError: java.lang.NoClassDefFoundError: java/lang/invoke/StringConcatFactory
	at me.drton.jmavsim.Visualizer3D$CustomCanvas3D.postRender(Visualizer3D.java:883)
	at javax.media.j3d.Renderer.doWork(Renderer.java:1376)
	at javax.media.j3d.J3dThread.run(J3dThread.java:271)
Caused by: java.lang.NoClassDefFoundError: java/lang/invoke/StringConcatFactory
	... 3 more
Error occurred during Canvas3D callback:
java.lang.BootstrapMethodError: java.lang.NoClassDefFoundError: java/lang/invoke/StringConcatFactory
	at me.drton.jmavsim.Visualizer3D$CustomCanvas3D.postRender(Visualizer3D.java:883)
	at javax.media.j3d.Renderer.doWork(Renderer.java:1376)
	at javax.media.j3d.J3dThread.run(J3dThread.java:271)
Caused by: java.lang.NoClassDefFoundError: java/lang/invoke/StringConcatFactory
	... 3 more
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:58)
Caused by: java.lang.BootstrapMethodError: java.lang.NoClassDefFoundError: java/lang/invoke/StringConcatFactory
	at me.drton.jmavlib.mavlink.MAVLinkMessageDefinition.calculateExtraCRC(MAVLinkMessageDefinition.java:46)
	at me.drton.jmavlib.mavlink.MAVLinkMessageDefinition.<init>(MAVLinkMessageDefinition.java:42)
	at me.drton.jmavlib.mavlink.MAVLinkSchema.processXMLFile(MAVLinkSchema.java:113)
Thank you!!

aryus96, 2017-10-18:
Hi !
I have the same problem.
Ubuntu 16.04 and Java 8 & 9 installed.
I can't remove Java 9 because CUDA 9.0 is using it, so do I.

I tried to erase everything and reinstall them, still same problem..

dr0i, 2018-01-04:
Don't use the `openjdk version` but the one provided by oracle. See e.g. https://tecadmin.net/install-oracle-java-8-ubuntu-via-ppa/ how to do this.

asimonov, 2018-01-23:
I see the same problem on Ubuntu 16.04 with ROS Kinetic. tried remove/install both openjdk and oracle jdk but see same error with both.

cosmicog, 2018-12-20:
Also after doing what @dr0i said, run:

    $ sudo apt remove openjdk-*
    $ sudo apt remove default-jre*
    <PX4 Firmware installation dir>/Tools/jMAVSim$ rm -rf out/

Open up a new terminal and go to `<PX4 Firmware installation dir>/Tools/jMAVSim` and 

    <PX4 Firmware installation dir>/Tools/jMAVSim$ ant clean distro

Now it should work.