PX4/PX4-Autopilot #4982: VTOL_LAND ignored

sanderux, 2016-07-04:
Mission on current master:
Kept loitering around waypoint 1/2, never proceeded to the landing waypoint.
Log: http://logs.uaventure.com/view/DwRTKb7jKf5LHXH553s6n7

Mission:
[image]

/cc @LorenzMeier @tumbili @AndreasAntener

sanderux, 2016-07-11:
@julianoes any help appreciated

dagar, 2016-07-11:
So it's stuck in a position setpoint after takeoff. The next time the setpoint is updated is RTL.
Here's the state of the code for this flight - https://github.com/PX4/Firmware/blob/d96810f250d60481193cf15b7e99cd49d5106028/src/modules/navigator/mission.cpp#L510

sanderux, 2016-07-11:
Considering it is not reproducible in sitl, could it be a memory issue?

sanderux, 2016-07-11:
I've been able to reproduce, it is likely due to coming out of transition inside the acceptance radius of the vtol_land command.

Steps to reproduce:
VT_ARSP_TRANS = 15
VT_TRANS_THR = 0.3
Load mission file: [VTOL_LAND loiter.mission.zip](https://github.com/PX4/Firmware/files/357993/VTOL_LAND.loiter.mission.zip)

Result:
[image]

sanderux, 2016-07-12:
Update: after more then 50 SITL flights i have only been able to reproduce this twice.
Both times the behavior appear directly after setting a complete new mission, however, many more times this did not result in the same behavior.

still looking into what could cause this behavior.

dagar, 2016-07-12:
Looking through the code it still seems like you're getting stuck here - https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission_block.cpp#L193

Did you get any additional information during your 2 reproduces? With the new logger you could add full logging of vehicle_status, fw_pos_ctrl_status, etc and reconstruct the acceptance radius navigator uses.

sanderux, 2016-07-12:
Both reproduces were on the same firmware used with the actual mission failure, i don't think advanced logging was part of that. I could enable it and try to reproduce again but i have failed the last 30+ times.

With no solid way to reproduce it will also be very hard to analyze / fix.

sanderux, 2016-07-12:
I will try to manipulate some values in the waypoint_reached code to see if i can manually trigger the anomaly and go from there.

dagar, 2016-07-12:
You can verify it's the same behavior by breaking that, but I'm wondering if there's some race condition where you're not getting the proper vstatus, or fw_pos_ctrl_status.

sanderux, 2016-07-13:
@julianoes i'm not getting much further on this, could you have a look?

julianoes, 2016-07-13:
@sanderux sorry I don't have bandwidth to look into this right now. I haven't played with any VTOL stuff so I'd have to work my way in.

I'm sure @tumbili can help much better here.

sanderux, 2016-07-23:
On current master i get consistent results that do not exhibit the same behavior but are likely related:
[image]

sanderux, 2016-07-26:
Another instance of a waypoint not reached: http://logs.uaventure.com/view/52Y2vFerC3bbEXdA5YvWVS

This time it was a regular waypoint, it circled it before reaching it before considering it reached

dagar, 2016-07-26:
Circling to hit the altitude?

sanderux, 2016-07-26:
could be, but on the initial reported error the altitude was never a problem.
It did occur directly after lowering thrust and thereby loosing a lot of altitude.

AndreasAntener, 2016-07-28:
@dagar we might have 2 issues here, but on master right now the default acceptance radius for Z is too little in the VTOL case. We set it to 3m for MC mode, so I have also 3m now in FW for Z. I guess we need a separate parameter for FW alt acc rad

dagar, 2016-07-28:
Alright, that's an easy fix, but it's unfortunate if we need duplicate parameters. Are there other instances like this?

LorenzMeier, 2016-07-29:
Closing this in favour of the PR.

closed 2016-07-29 by LorenzMeier