PX4/PX4-Autopilot #4983: ppm issue

alexFab, 2016-07-04:
i try to get openLRS with PX4 to work, but no success so far. I found a [post ](http://diydrones.com/forum/topics/apm-ppm-from-openlrsng?commentId=705844%3AComment%3A1737913)where someone describes the problem.

> Actually OpenLRS produces a very short 'low' pulse on the serial PPM, like 50 µs. For some systems this is not an issue, as the important data is the time between two pulses, not the pulses themselves.
> But for APM (and Pixhawk), this is too short. [....] normal 300 µs low pulses and this works fine

Is it possible to fix this in the current version? or is this issue known?

alexFab, 2016-07-05:
the main developer of openLRS pointed out that this Timing issue is fixed log time ago. so this is no longer the Problem. [see](http://www.rcgroups.com/forums/showpost.php?p=35179054&postcount=8554)

alexFab, 2016-07-06:
px4 doesn't recognize my openLRS RX (submicroLRS). But [this](https://storage.googleapis.com/google-code-archive-downloads/v2/code.google.com/generate-ppm-signal/Generate_PPM_signal_nodelay.ino) simple ppm simulator works fine with px4. 
ppm signal from openLRS looks pretty much the same like ppm from simulator but inverted. 

this issue is also open @openLRS openLRSng/openLRSng#113

LorenzMeier, 2016-07-06:
Can you attach plots and data from your logic analyser? We support non-inverted and inverted PPM.

alexFab, 2016-07-06:
openlrs - doesn't work with px4
[image]

simulator - works with px4
[image]

LorenzMeier, 2016-07-06:
Can you please show 2 cycles / bursts? The pauses between the signals also do matter.

alexFab, 2016-07-06:
not at the moment, maybe later today. but the pauses between the signals are configurable in openLRS ([Minimum PPM sync time](https://github.com/openLRSng/openLRSngWiki/wiki/Settings-Guide#receiver-settings)). on the screenshots the pauses are something like 4300us. I did also tests with 3000us.

alexFab, 2016-07-06:
from: ([Minimum PPM sync time](https://github.com/openLRSng/openLRSngWiki/wiki/Settings-Guide#receiver-settings))

> The minimum time between PPM frames on the PPM output. The default value works fine for many systems but fickle systems such as Ardupilot require a longer sync time (ardupilot 3.1 and better works ok with 3000, some older versions apparently want 4200).

alexFab, 2016-07-06:
did some measurements again. and realized there are different offsets on rc-in. the screenshots above are made while the devices are not connected, so without the offset voltage, things looks different if the devices are connected.  After rebooting, before rc-in sees a vaild ppm-signal it has this Signal:

[image]

rc-in with connected openLRS  it looks like this:
[image]

rc-in with connected ppm simulator (detected by px4),
[image]

rc-in without any connection. the offset has changed now to 3.3V-DC
[image]

rc-in with the 3.3V-DC offset and connected openLRS receiver.
[image]

openLRS standalone
[image]

simulator standalone
[image]

alexFab, 2016-07-06:
I also try apm on pixRacer. No more offset voltage on the rc-pin but in the moment i connect the ppm pin from openLRS with the rc-in pin the signal max voltage drops from 3.3V to 1V. I guess this is to low.

alexFab, 2016-07-07:
finally i was able to get openLRS with px4 on pixRacer to work.

openLRS produces a ppm signal Upp 3.3V at least the submicroLRS receiver. But px4 needs ppm Upp 5V. 

here my hardware solution, but i would prefer the software solution on PX4 side ;)

[image]

LorenzMeier, 2016-07-07:
That conclusion doesn't seem to be correct. All the hardware we've using outputs 3.3V. Did you by chance invert the signal with this schematic?

alexFab, 2016-07-07:
yes, signal inversion is a side affect of this logic level converter.

gregd72002, 2016-10-30:
Hi, I stumbled upon the same issue. Is there a way to get Pixracer accept inverted PPM signal as RC-in?

LorenzMeier, 2016-12-26:
Closing since stale.

closed 2016-12-26 by LorenzMeier

Deca, 2017-02-19:
I've the same issue, voltage drop on ppm when connected to rcin and the pixracer don't receive any signal, anyone found a solution?

t413, 2017-05-30:
Thank you @alexFab! I made a N3904 inverter as you suggested and it worked like a charm. CPPM from my OpenLRS receiver works great. Here's what my inverter looks like: (pre heat shrink)

[image]

Siytek, 2017-08-07:
So just to confirm, PX4 does not support inverted PPM? TBS Crossfire also outputs an inverted PPM, which is not recognised by PX4.

No problem to make a little hardware inverter if necessary but a software solution would avoid adding the extra point of failure to the system.

LorenzMeier, 2017-08-08:
It does support it. I would need to review why the decoder is rejecting it. From a HW perspective it should all work. @Siytek do you have the embedded software skills required to inspect the decoder?

https://github.com/PX4/Firmware/blob/4453e4201b7a245cff52beeb38a293161aea4c48/src/modules/systemlib/ppm_decode.c

Siytek, 2017-08-08:
Hi Lorenz, I am up for taking a look at it although I will not claim to have the skills! I think the first thing to do is to scope my PPM signal and make sure it conforms to the parameters given in ppm_decode.c - I will get this done and post the results as soon as I get chance.

LorenzMeier, 2017-08-08:
I've updated the link to where the decoding is actually happening: https://github.com/PX4/Firmware/blob/master/src/drivers/stm32/drv_hrt.c#L448-L601

Siytek, 2017-08-08:
Where is the value for `ppm.next_channel` obtained? The only place I could find a value for it was when it is reset on line 520 in drv_hrt.c

Siytek, 2017-08-09:
Also when the state machine is ACTIVE it doesn't seem to do anything with the calculated interval value? Isn't it supposed to store it in the ppm buffer for usage at this point?

Siytek, 2017-08-09:
My PPM on the scope:

[image]

Starting pulse = 9.36ms
Channels ≈ 1.2ms

AlexFa, 2017-08-11:
can someone confirm that px4 handles inverted PPM now correctly? Without a hardware-inverter?

Siytek, 2017-08-11:
Hi @alexFab, I was being stupid with my wiring and now I have put it right all is working fine with PPM. As for inverted PPM, your signal shown in your screen shot 'OpenLRS alone' looks the same as mine (not inverted). If you are running 8 channels, you can see in your screen shot 8 short peaks for each channel and a long peak for the start signal. Its a little misleading as there is so little time spent in the low state, its easy to mistake it for inverted!

[image]

Are you sure your circuit is not acting as some kind of buffer, which could be correcting the problem, rather than the fact that it inverts the signal? I used to use Orange OpenLRS before I got my TBS Crossfire, I don't remember entirely but I think I used to use it with PPM and I never had any problems.