PX4/PX4-Autopilot #4984: Oscillation occurs when switch from Follow Me to Position

whyscience, 2016-07-05:
Hi 
 I fly in Follow Me mode ,when I switch to Position mode ,there will be oscillation in attitude like this 
[image]

I test the latest firmware and V1.3 ,they have the same problem .
Anyone could help me with it ?

dagar, 2017-04-12:
@Stifael would the mc position controller handle this better now?

Stifael, 2017-04-12:
@dagar. I dont think so, did not get to it to look into follow me and how it is implemented.

dagar, 2017-04-28:
@whyscience have you tried this recently? There's been a lot of work on the MC position controller.

TSC21, 2017-08-20:
Stale

closed 2017-08-20 by TSC21