PX4/PX4-Autopilot #4985: HIL not working

vvranjek, 2016-07-05:
On the latest master (dc1f343) HIL simulation with X-Plane stopped working compared to the master from a few weeks back. When I try to switch to mission mode it says: "Set mode command temporarily rejected". I also had to turn off RC link loss failsafe, otherwise it would go straight to failsafe mode.

LorenzMeier, 2016-07-05:
@dagar Would you mind reconfirming?

dagar, 2016-07-05:
I was using HIL heavily yesterday with only some navigator changes (LOITER_TO_ALT) on top of this commit.
@vvranjek is it possible you tried before you had a home position? With EKF2 that isn't shown in the mavlink messages, but I had a console connected to serial 5.

dagar, 2016-07-05:
Ok I just saw some kind failure where I connected HIL to xplane and it went straight into failsafe that I can't escape. Must be some kind of race condition.

dagar, 2016-07-05:
We really need better error messages. What caused failsafe mode, why was a mode change temporarily rejected, etc.

dagar, 2016-07-05:
If you arm before getting a home position you'll never get one.
https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L2717

LorenzMeier, 2016-07-05:
That's intended. What we need is to probably disallow this by default when we have a GPS module detected. This way we don't harass racing pilots but make sure camera ships RTL correctly.

vvranjek, 2016-07-06:
@dagar I don't think I get the home position. After I connect to the simulator, I get these 3 messages in console:
```
WARN  [navigator] navigator: global position timeout
INFO  [lib__ecl] EKF using pressure height - commencing alignment
INFO  [lib__ecl] EKF alignment complete
WARN  [mavlink]  Disabling hardware flow control
```

dagar, 2016-07-06:
Can you reset the pixhawk, reconnect QGC/XPlane and wait at least a minute to see if you ever get it?

vvranjek, 2016-07-06:
Nope, I don't even see GLOBAL_POSITION_INT in mavlink inspector but GPS_RAW_INT is there. Is this only my problem or has anyone confirmed this?

dagar, 2016-07-06:
Have you had HIL with QGC and XPlane working before? It's probably worth verifying your XPlane settings either way. https://pixhawk.org/users/hil

Once you connect does GPS_RAW_INT show the position of the plane in XPlane? 

In the console can you run `commander status`

vvranjek, 2016-07-06:
Yeah, I had it working from a modified master commit from a few weeks ago. I merge master every few weeks and thats what broke it for me. I tried a clean master and it's the same thing. 

Yes, GPS_RAW_INT shows the position.

I just realized that I also have a bootloader problem. Below is the boot log as well as commander status after connecting to the sim and waiting about a minute. I tried reflashing several times to make sure the bootloader problem persists.
```
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
WARN  [param]    selected parameter default file /fs/mtd_params
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
WARN  [px4io]    check CRC failed: -22
INFO  [px4io]    default PWM output device
[PX4IO] using firmware from /etc/extras/px4io-v2.bin
[PX4IO] bad sync 0xff,0xdf
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0x7f,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0x7f
[PX4IO] bad sync 0xff,0xff
[PX4IO] bad sync 0xff,0xff
[PX4IO] bootloader not responding
WARN  [px4io]    check CRC failed: -22
WARN  [param]    Error: Parameter TRIG_MODE not found.
INFO  [dataman]  Power on restart, data manager file '/fs/microsd/dataman' size is 103090 bytes
MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
WARN  [bst]      no devices found
INFO  [ver]      match: PX4FMU_V2
WARN  [hmc5883]  no device on bus 2
read_reg failWARN  [lis3mdl]  no device on bus 2
WARN  [hmc5883]  no device on bus 1
ERROR [mpu6000]  no device on this bus
MPU6000 on SPI bus 1 at 4 (1000 KHz)
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
ERROR [meas_airspeed] no MS4525 airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
nsh: sf10a: command not found
INFO  [pwm_out_sim] MODE_8PWM
INFO  [mavlink]  mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [mavlink]  mode: OSD, data rate: 1000 B/s on /dev/ttyS2 @ 57600B
WARN  [param]    Error: Parameter TRIG_MODE not found.
INFO  [ver]      match: PX4FMU_V2
nsh: px4flow: command not found
px4flow [174:100]
INFO  [mavlink]  mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
INFO  [init] Fixedwing 
INFO  [init] Mixer: /etc/mixers/AERT.main.mix on /dev/pwm_output0 
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1 

INFO  [init] Addons script: /fs/microsd/etc/extras.txt 

NuttShell (NSH)
nsh> \0x1b[KWARN  [mavlink]  Disabling hardware flow control
WARN  [mavlink]  Disabling hardware flow control
WARN  [navigator] navigator: global position timeout
INFO  [lib__ecl] EKF using pressure height - commencing alignment
INFO  [lib__ecl] EKF alignment complete
commander status
WARN  [commander] type: forward motion
WARN  [commander] safety: USB enabled: [OK], power state valid:  [OK]
WARN  [commander] avionics rail:   4.61 V
WARN  [commander] home: lat = 0.0000000, lon = 0.0000000, alt = 0.00, yaw: 0.00
WARN  [commander] home: x = 0.0000000, y = 0.0000000, z = 0.00 
WARN  [commander] datalink: OK
WARN  [commander] arming: STANDBY
nsh>
```

dagar, 2016-07-06:
First let's get the IO flashed, in case it's blocking something.

    px4io forceupdate 14662 /etc/extras/px4io-v2.bin

vvranjek, 2016-07-07:
Ok, updated successfully, no more errors, but everything else seems the same. Here is the boot log with commander status:
```
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
WARN  [param]    selected parameter default file /fs/mtd_params
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
WARN  [px4io]    CRCs match
WARN  [param]    Error: Parameter TRIG_MODE not found.
INFO  [dataman]  Power on restart, data manager file '/fs/microsd/dataman' size is 103090 bytes
MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
WARN  [bst]      no devices found
INFO  [ver]      match: PX4FMU_V2
WARN  [hmc5883]  no device on bus 2
read_reg failWARN  [lis3mdl]  no device on bus 2
WARN  [hmc5883]  no device on bus 1
ERROR [mpu6000]  no device on this bus
MPU6000 on SPI bus 1 at 4 (1000 KHz)
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
ERROR [meas_airspeed] no MS4525 airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
nsh: sf10a: command not found
INFO  [pwm_out_sim] MODE_8PWM
INFO  [mavlink]  mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [mavlink]  mode: OSD, data rate: 1000 B/s on /dev/ttyS2 @ 57600B
WARN  [param]    Error: Parameter TRIG_MODE not found.
INFO  [ver]      match: PX4FMU_V2
nsh: px4flow: command not found
px4flow [169:100]
INFO  [mavlink]  mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
INFO  [init] Fixedwing 
INFO  [init] Mixer: /etc/mixers/AERT.main.mix on /dev/pwm_output0 
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1 

INFO  [init] Addons script: /fs/microsd/etc/extras.txt 

NuttShell (NSH)
nsh> \0x1b[KWARN  [mavlink]  Disabling hardware flow control
WARN  [mavlink]  Disabling hardware flow control
WARN  [navigator] navigator: global position timeout
INFO  [lib__ecl] EKF using pressure height - commencing alignment
INFO  [lib__ecl] EKF alignment complete

nsh> \0x1b[Kcommander status
WARN  [commander] type: forward motion
WARN  [commander] safety: USB enabled: [OK], power state valid:  [OK]
WARN  [commander] avionics rail:   4.64 V
WARN  [commander] home: lat = 0.0000000, lon = 0.0000000, alt = 0.00, yaw: 0.00
WARN  [commander] home: x = 0.0000000, y = 0.0000000, z = 0.00 
WARN  [commander] datalink: OK
WARN  [commander] arming: STANDBY
nsh>
```

dagar, 2016-07-07:
Something else must be different. I think we have nearly identical setups, but mine is still working consistently.
Which airframe do you have selected (SYS_AUTOSTART)?
Can you verify HIL still works if you flash PX4 stable?

vvranjek, 2016-07-07:
I just tried the latest master (2317589) and it works now, but (dc1f343) from few days ago is broken for me. I'll just stick to the newer version for now. Thanks anyway!

dagar, 2016-07-07:
Unsatisfying conclusion, but I'm glad it's now working for you. ECL was updated in this time frame.

https://github.com/PX4/Firmware/compare/dc1f343...2317589

closed 2016-07-07 by dagar