PX4/PX4-Autopilot #4988: Python module & scripts to convert ULog files

bkueng, 2016-07-05:
The following repo contains scripts for ULog files: https://github.com/PX4/pyulog

Among other things it contains a CSV conversion script.
@kd0aij fyi

julianoes, 2016-07-05:
@bkueng Nice work!

kd0aij, 2016-07-05:
@bkueng Thanks, the scripts work for me.
But I noticed that parameter changes are no longer being logged as described here: https://github.com/PX4/Firmware/pull/4400#issuecomment-216083028
Was that intentionally removed, or just lost?

bkueng, 2016-07-06:
@kd0aij no that's a regression. But I just tested, it still works for me in SITL, I can see the changes in FlightPlot and with `params.py`.

kd0aij, 2016-07-06:
Sorry, I was confused: From looking at multiple logs it is obvious that I changed the parameter while disarmed :)

bkueng, 2016-07-06:
good to hear!

closed 2016-07-06 by bkueng