PX4/PX4-Autopilot #4995: Test multirotor control without mixer limits

LorenzMeier, 2016-07-06:
@MaEtUgR Medium term interested in LQR / MPC, but as first step I'd like to work out if the mixer limiting is negatively impacting performance.

LorenzMeier, 2016-12-28:
@MaEtUgR Found this paper by chance today:
https://www.ifi.uzh.ch/dam/jcr:5f3668fe-1d4e-4c2b-a190-8f5608f40cf3/RAL16_Faessler.pdf

It combines a naive mixer with its implementation. Its not clear how much different their work is to what we already have, but its possible there are things left to improve in our not-completely-naive mixer.

kd0aij, 2016-12-28:
@tumbili The paper referenced above also has a discussion of motor thrust/torque modeling which may be of interest to you. They chose to use all 3 coefficients to fit a quadratic polynomial to the measured thrust.

MaEtUgR, 2016-12-28:
@LorenzMeier Thank you for the reference. We talked about the mixer in a control call and we have an idea how to improve the mixer. Improving the rate controller will lead sooner or later to the mixer priorities and I'll consider the results of the paper for this.

closed 2018-02-21 by PX4BuildBot

MaEtUgR, 2018-02-21:
There's actually progress: #7920
And I just discussed with @bkueng that I could still add functionality on top like yaw priorization from the draft mixer I pushed here before: #8485