PX4/PX4-Autopilot #5001: Add ability to set target altitude atomically, optionally by an offset

superware, 2016-07-07:
It is sometimes required to change altitude with the following requirements:

1. Change target altitude by an offset, so the GCS isn't required to "know" current target altitude (in the condition of down-link loss, for example).
2. Set target altitude only, without affecting target position. MAV_CMD_DO_REPOSITION seems to set target position to current position for NaN Latitude and Longitude.
3. Set relative target altitude. For example, set target altitude to 100m above home position. Currently this requires the GCS to relay on HOME_POSITION.altitude or calculate (GLOBAL_POSITION_INT.alt - GLOBAL_POSITION_INT.relative_alt), but home position can change at any time, making this a bit difficult to get right at all times.

Maybe MAV_CMD_NAV_CONTINUE_AND_CHANGE_ALT is related to some of the above.

LorenzMeier, 2016-07-07:
@dagar should we add this to your PR?

dagar, 2016-07-07:
Hi @superware 
I proposed adding MAV_CMD_DO_CHANGE_ALTITUDE to mavlink common, but we couldn't come to an agreement. https://github.com/PX4/Firmware/issues/5001#issuecomment-231022023
MAV_CMD_NAV_CONTINUE_AND_CHANGE_ALT, MAV_CMD_CONDITION_CHANGE_ALT, and MAV_CMD_DO_REPOSITION didn't seem quite right to me.

                <entry value="186" name="MAV_CMD_DO_CHANGE_ALTITUDE">
                    <description>Change altitude set point.</description>
                    <param index="1">Altitude type (0=amsl, 1=local, 2=relative, 3=terrain)</param>
                    <param index="2">Altitude in meters</param>

On the PX4 side the idea was that this message could set the altitude for altitude/position control modes, or temporarily override the mission altitude. 

Given your interest let's make sure this message is appropriate for what you're trying to accomplish and then try to get accepted in mavlink.

ecmnet, 2016-07-07:
Maybe MAV_CMD_NAV_TAKEOFF should be included into that discussion, as it takes currently asl as parameter only.

dagar, 2016-07-07:
What about using COMMAND_INT and specifying the MAV_FRAME?
Small change here to adjust the altitude according to frame limited to appropriate commands.
https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L449

superware, 2016-07-07:
@dangar MAV_CMD_DO_CHANGE_ALTITUDE sounds perfect from my side.

I would also add an "offset" altitude-type, so for example a (gamepad) button can be a really quick 1m climb/descend without the latency of relaying on incoming periodic messages (to get current target altitude etc), meaning three clicks in a row will actually climb 3m, and not climb 1m for three messages containing the same absolute altitude. Also, when executing missions, I can think of a few scenarios which require offsetting altitude.

`<entry value="186" name="MAV_CMD_DO_CHANGE_ALTITUDE">
                <description>Change altitude set point.</description>
                <param index="1">Altitude type (0=amsl, 1=local, 2=relative, 3=terrain, 4=offset)</param>
                <param index="2">Altitude in meters</param>`

Hacks and workarounds are always possible, but a generic "change altitude" command that will take into consideration all possible/future requirements should be optimal.

@ecmnet IMHO you are right, actually this was the first issue I encountered when started with PX4/MavLink just a few weeks ago:

'<entry value="22" name="MAV_CMD_NAV_TAKEOFF">
                <description>Takeoff from ground / hand</description>
                <param index="1">Minimum pitch (if airspeed sensor present), desired pitch without sensor</param>
                <param index="2">Altitude type (0=amsl, 1=local, 2=relative)</param>
                <param index="3">Empty</param>
                <param index="4">Yaw angle (if magnetometer present), ignored without magnetometer</param>
                <param index="5">Latitude</param>
                <param index="6">Longitude</param>
                <param index="7">Altitude</param>
</entry>'

dagar, 2016-07-07:
Actually I think we should just be using MAV_FRAME which has everything we've mentioned and more.
http://mavlink.org/messages/common

You can't change MAV_CMD_NAV_TAKEOFF, but you could specify the MAV_FRAME with COMMAND_INT.

dagar, 2016-07-07:
There's also MAV_CMD_NAV_TAKEOFF_LOCAL.

@superware I'll reopen the MAV_CMD_DO_CHANGE_ALTITUDE pull request and we can continue the discussion over there.

superware, 2016-07-07:
Actually COMMAND_INT sounds good, is it already implemented in any way?

dagar, 2016-07-08:
For takeoff you'd need to explicitly handle it in mavlink_receiver and depending on frame type save the altitude differently. 

Relative is altitude above home. Local is the vehicle.

How were you imagining MAV_CMD_DO_CHANGE_ALTITUDE would work running a mission? I could see it overriding the altitude setpoint once, but then does the next waypoint set the altitude? I'll open the basic PR, but we should work through the details of it's intended usage before it's merged.

dagar, 2016-07-08:
@superware MAV_CMD_DO_CHANGE_ALTITUDE is now in mavlink. Are you interested in doing the PX4 implementation?

superware, 2016-07-08:
I'm just starting with the PX4 ecosystem, implementing anything is currently beyond me, unfortunately.

But I'm a bit confused, you said:

> Actually I think we should just be using MAV_FRAME which has everything we've mentioned and more.

dagar, 2016-07-08:
MAV_FRAME in the mavlink spec
http://mavlink.org/messages/common

[image]

kd0aij, 2016-07-08:
It looks like the descriptions for LOCAL_NED, and LOCAL_ENU are confused.
Also, I find the term "body NED frame" to be confusing. "North" and "East" are defined in an earth-fixed frame, not a body frame.

dagar, 2016-07-08:
Care to fix it?
https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml#L474

kd0aij, 2016-07-08:
Since everything I know about MAVLINK comes from this documentation, I'm not actually qualified to change it, only to express confusion at its meaning, and speculate as to intent :)

TSC21, 2018-02-21:
@dagar is this still relevant?

superware, 2018-02-27:
@dagar IMHO this is still quite relevant. Is there a current way to increase altitude by an offset? (relative to the frame)

anandbaskaran, 2018-03-12:
It seems there was lots of yes and no for this implementation. But can you give some light on the final situation of this? 

I would like to change the altitude of the drone dynamically (by sending Mavlink message) during the mission mode.

kfruson, 2018-06-26:
Any updates on being able to change the altitude during a mission?

anandbaskaran, 2019-01-30:
It would be great to have this option! @bot please don't close it :)

closed 2019-02-13 by stale[bot]