PX4/PX4-Autopilot #5002: Request for VTOL logs

sanderux, 2016-07-07:
In an attempt to hunt down a few recent issues i would like to get some logs from VTOL flights that include transitions and preferably missions. Please indicate if your mission was using VTOL_TAKEOFF or regular takeoff and transition, and what firmware version you were running.
Manual flights also welcome.

Any additional info like wind conditions and airplane layout is welcome.

@tumbili @AndreasAntener @LorenzMeier @SimonWilks @tubeme @CarlOlsson @nkhoit

CarlOlsson, 2016-07-07:
I don't know how useful it is since we are not on upstream px4 but here is a flight with our platform
http://logs.uaventure.com/view/5sFcbxJH7oJ35TcFVbrE2C
sunny and little wind, regular takeoff

sanderux, 2016-07-07:
From @tikiti
http://logs.uaventure.com/view/Mqpza4a78VJTTSXQ7kXiLQ
FireFly6 with Pixhawk
Full pilot-control
Hover/VTOL take-off and landing, forward flight transition at around 50M. Tested all flight modes (Man-ACRO, Alti, Position, Loiter)
Return and transition back in manual mode at about 50M again.
Manual-STAB landing

closed 2016-07-13 by sanderux