PX4/PX4-Autopilot #5008: Fix MAVROS rotations

LorenzMeier, 2016-07-07:
@ChristophTobler @simonegu @vilhjalmur89 @julianoes We will need to inspect the MAVROS transformations. We should look at this and discuss it tomorrow.

Associated MAVROS issue: https://github.com/mavlink/mavros/issues/523

LorenzMeier, 2016-07-12:
@julianoes http://www.ros.org/reps/rep-0103.html

The key sentence is: yaw 0 is facing east, so when converting from NED to ENU a 0 degree heading will turn into a 90 degree heading (since the rotation direction around yaw is inverted as well, due to the inverted Z axis).

julianoes, 2016-07-13:
From what @vilhjalmur89 and I found it seems that we have to work in different coordinate frames in PX4, ROS, gazebo but the transformations in MAVROS seem to be implemented correctly.

PX4: NED
MAVROS: ENU
Gazebo: NWU (although it's not sure if Gazebo has an understing of North, or rather just x,y,z)

LorenzMeier, 2016-07-13:
I would like to see the reference that states that ROS is NWU. Please re-read REP 103 which says 0 yaw is facing east.

julianoes, 2016-07-14:
> I would like to see the reference that states that ROS is NWU. Please re-read REP 103 which says 0 yaw is facing east.

I don't understand, it says ROS is ENU in http://www.ros.org/reps/rep-0103.html#axis-orientation
and that matches with yaw 0 pointing to East.

NCharabaruk, 2016-11-14:
Has this been looked into any further?

julianoes, 2016-11-14:
I still think this is all correct. I'm waiting for somebody showing me proof of the opposite :)

NCharabaruk, 2016-11-14:
So it's normal for the yaw being shown by /mavros/local_position/pose to be 180deg off from that being published to /mavros/vision_pose/pose?

julianoes, 2016-11-14:
I don't know what's normal :) but if you point me to a field which is not according to the specs I'll believe it. So please dig out the specs for the mavros fields that you cite and we can check if it's correct.

closed 2016-11-19 by LorenzMeier