PX4/PX4-Autopilot #5013: Problem connecting to QGC after uploading custom firmware version 1.3.4

mohit004, 2016-07-09:
Dear PX4 Developers,

I am trying to create new mixer files for my experimental setup. In order to do that, I tried putting new mixer and config files in 'ROMFS/px4fmu_common/mixers' and 'ROMFS/px4fmu_common/init.d', respectively. After building the code using 'make px4fmu-v2_default', everything goes well and the file 'airframes.xml' is created and it even includes my new configuration. Subsequently, I do 'make px4fmu-v2_default upload' and the new firmware is successfully uploaded onto pixhawk board. However, when I connect to QGC, I am facing two issues:

1. **My custom airframe does not appear in the airframe list**, as I expected. Even after trying another approach of including the mixer and config file in the microsd card at '/etc/mixers' and '/etc/init.d', respectively, as specified in http://dev.px4.io/advanced-system-startup.html, my custom airframe does not appear in the current airframe list. To make sure that the pixhawk is taking the mixer files from the sd card while booting, I tried modifying a current airframe configuration and then including it in the sd card with the same name. After selecting that config from QGC, I can see my custom mixer is working fine. 

2. **Another issue that I am facing is with the connection with QGC after uploading the custom firmware itself**. After I upload the firmware on pixhawk, I connect to QGC and it (QGC) is able to detect the firmware and gets connected. However, after I select the airframe and reboot the pixhawk (as required by QGC), QGC is not able to connect to pixhawk at all. I tried changing lot of things like baudrate, etc., but it still does not recognise the pixhawk board. Moreover, with pixhawk everything seems fine as it boots with flashing blue light and later red light starts blinking, which I suppose is due to sensors not calibrated. I tried searching online for some solutions but nothing was available. In order to make sure that there is nothing wrong with the current firmware version (v1.3.4) I am using, I tried using previous versions like v1.3.1 and v1.3.3 and I did not face this problem with them. Kindly confirm if there is any problem with the firmware version v1.3.4 or am I missing something?

I am using Ubuntu 14.04, QGC v2.9.7b-417-g627025d and latest version of pixhawk hardware (http://dev.px4.io/hardware-pixhawk.html).

Any of your guidance is greatly appreciated.

Kind Regards
Mohit

dagar, 2016-07-09:
You need to use QGC to flash your custom firmware for it to get a copy of the metadata.

mohit004, 2016-07-12:
Dear Dagar,

Thanks a lot for your response. I tried it and thankfully, now I am able to see my custom config in the airframe list.
However, as before, after I upload the custom firmware and select my airframe config, the pixhawk doesn’t connect with QGC. I have tried using firmware version 1.3.4 and 1.4.1 but the same thing is happening with both of them.  it gives me an error:
'your vehicle is not responding. If this continues shutdown QGC, restart the Vehicle letting it boot completely, then start QGC.'. I restart and wait for it to boot and then start QGC but no success.

When I tried with firmware v-1.3.3 and v-1.3.1, it (pixhawk) connects with QGC even after selecting the airframe but shows some warnings like 
'old firmware is being used'
'system boot incomplete'

All I am doing to put my custom configuration is including a config file (14003_...) in 'ROMFS/px4fmu_common/init.d' and my mixer file in 'ROMFS/px4fmu_common/mixers'. I dont think that there is something wrong with my mixer file as I am able to use is with the generic airframes (already included in PX4 flightstack). Furthermore, I thing my config is also correct as I almost copied one thats already included. Just to make sure, the content of my config file is
_sh /etc/init.d/rc.mc_defaults
set MIXER tri_y_yaw_roll+
set PWM_OUT 12345_

I think, maybe I am missing to run some startup files like 'rcS', etc. included in 'init.d' folder. Do we need to call these startup files explicitly when we add a custom configuration.

Thanks once again for your help.

Kind Regards
Mohit

dagar, 2016-07-12:
Is your sdcard in?
As long as you didn't modify the existing rcS, rc.mc_defaults, etc the custom stuff you added should be fine.

kd0aij, 2016-07-12:
Do you have a terminal connected to the nsh console? There might be some
boot messages going to the console that would be informative.

mohit004, 2016-07-12:
Dear Dagar,

My Sdcard is perfectly inside.
Yes, I didnot modify any of these custom startup files.

 Dear kd0aij,

I have not connected terminal to nsh console. But I can try that. Thank you for your suggestion.

Kind Regards
Mohit

dagar, 2016-07-12:
Another quick thing to check would be flashing regular PX4 1.4.1 through QGC and making that sure that connects.

mohit004, 2016-07-12:
Dear Dagar,

Yes, I already checked it 5-10 times, the regular firmware (all versions) work perfectly. Even my custom Firmware (v-1.4.1) works fine as long as I select the generic airframe and not my custom one.

Thanks

dagar, 2016-07-12:
Oh, in the case I retract my previous statement. It sounds like there's some issue in one of your files.

If you connect to the system console (http://dev.px4.io/advanced-system-console.html) you might get a better error message. Otherwise share your airframe, mixer, etc and we'll look for the problem.

mohit004, 2016-07-12:
Dear Dagar,

Actually, in the beginning I started by modifying the already existing tricopter configuration by adding a servo to roll input. Below you may find my mixer and config files.

**Mixer file**

# Tricopter Y+Configuration Mixer
# Yaw Servo +Output ==> +Yaw Vehicle Rotation
# Roll Servo +Output ==> +Roll Vehicle Rotation

# Motors
R: 3y 10000 10000 10000 0

# Yaw Servo
M: 1
O:      10000  10000      0 -10000  10000
S: 0 2  10000  10000      0 -10000  10000

# Roll Servo
M: 1
O:      10000  10000      0 -10000  10000
S: 0 0  10000  10000      0 -10000  10000

**Config file**

sh /etc/init.d/rc.mc_defaults
set MIXER tri_y_yaw_roll+
set PWM_OUT 12345

Furthermore, I have not tried connecting to console yet. Now only I am making connections so that I can access console using FTDI cable. I will update the error msg soon.

Kind Regards
Mohit

dagar, 2016-07-12:
Nothing obviously wrong. Sometimes even white space errors can cause problems. Watching the console during boot will tell us what's failing.

mohit004, 2016-07-12:
Ohhh. In that case, I try to get the error msg from console soon.

mohit004, 2016-07-12:
Hi Dagar,

this is the message stream that I got!

_MPU6000 on SPI bus 1 at 4 (1000 KHz)
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
ERROR [meas_airspeed] no MS4525 airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
ERROR [ets_airspeed] no ETS airspeed sensor connected
nsh: sf10a: command not found
INFO  [px4io] default PWM output device
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [mavlink] mode: OSD, data rate: 1000 B/s on /dev/ttyS2 @ 57600B
INFO  [ver] match: PX4FMU_V2
px4flow [172:100]
WARN  [px4flow] scanning I2C buses for device..
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
INFO  [init] Multicopter 
WARN  [init] Unknown MAV_TYPE 
INFO  [init] Mixer: /etc/mixers/tri_y_yaw_roll+.main.mix on /dev/pwm_output0 
ERROR [pwm] PWM_SERVO_SET_SELECT_UPDATE_RATE
ERROR [pwm] Task exited with errno=22_

Thankyou
Mohit

mohit004, 2016-07-12:
But as I know, this (PWM_RATE 400) is the default that has been used for Generic Tricopter configuration that already includes a servo (which works perfectly fine) and all I am doing is just including one more.

kd0aij, 2016-07-12:
have you tried changing your config file from "set PWM_OUT 12345" to "set
PWM_OUT 1234" ?

mohit004, 2016-07-12:
Not yet! Just give me a moment.

mohit004, 2016-07-12:
Dear Mark Whitehorn,

Yes, it works now! Thanks a lot.
Could you also kindly explain it to me that why did it work like that. Because what I thought while writing this config file was since I have 5 outputs, I must have PWM signals in 5 channels. Isn't is correct?

Thanks once again for your help!

Mohit

dagar, 2016-07-12:
https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rc.interface#L74

Now it's only setting the rate for 1-4, and leaving 5 alone (default).

mohit004, 2016-07-12:
Ohhh, but what if in an airframe configuration, I have 2 motors and 4 servos? wouldn't the config file end like 

...
_PWM_OUT 123456_

Moreover, if we dont need to open the output channels by specifying 'PWM_OUT <num_of_output_channels>' in the config file, how does the pixhawk know how many channels to output? Does it directly get from the number of outputs we define in mixer file (R: or M:)?

Thankyou
Mohit

kd0aij, 2016-07-12:
@mohit004 I think the only way to answer your open questions is to carefully read and analyze the source code. All I did was to grep for "set PWM_OUT" and observe that "12345" is used nowhere else. "123456" didn't show up either, but "12345678" is used in a few configs. If you have a digital tail servo and want to run it at 400Hz, I'd suggest trying that.

mohit004, 2016-07-12:
Dear Mark Whitehorn,

Thanks a lot for your response. I have been trying to read the source code for some time but I think I still dont understand it fully. Anyways, I think I will be able to get more comfortable with it slowly, in some more time. Actually, my aim is to develop a VTOL UAV (a bicopter configuration) with two motors and 4 servos (2 for differential tilting and other 2 for inward tilling). I will definitely try your suggestions. 

Thanks a lot once again.

Kind Regards
Mohit
P.S. Thanks a lot @dagar for your help too!

LorenzMeier, 2016-12-28:
Closing as stale.

closed 2016-12-28 by LorenzMeier