PX4/PX4-Autopilot #5028: MavLink unstable

vvranjek, 2016-07-11:
I had a few issues with mavlink lately. I have a custom mavlink message called CAP_STATUS that I configured to be sent every second. It was working fine until WIND_COV got implemented. After that I suddenly stopped receiving the message (I would randomly receive it every minute or so) and couldn't figure out why. The `mavlink_msg_cap_status_send_struct(_mavlink->get_channel(), &msg);' was executed every second, but the message just wasn't sent/delivered. Then I realized that if I change the send rate to anything but 1.0,  everything worked fine. I tried 1.1, 0.9, 5.0, 10.0, everything works as expected. I also tried changing other rates, that were set to 1.0 seconds with positive results. Here are a few working and non-working configurations:

Not working:
```
                configure_stream("SYS_STATUS", 1.0f);
		configure_stream("ATTITUDE", 15.0f);
		configure_stream("GPS_RAW_INT", 3.0f);
		configure_stream("GLOBAL_POSITION_INT", 3.0f);
		configure_stream("CAP_STATUS", 1.0f);
		configure_stream("VFR_HUD", 2.0f);
		configure_stream("EXTENDED_SYS_STATE", 1.0f);
		configure_stream("WIND_COV", 1.1f);
```
Working:
```
                configure_stream("SYS_STATUS", 1.1f);
		configure_stream("ATTITUDE", 15.0f);
		configure_stream("GPS_RAW_INT", 3.0f);
		configure_stream("GLOBAL_POSITION_INT", 3.0f);
		configure_stream("CAP_STATUS", 1.0f);
		configure_stream("VFR_HUD", 2.0f);
		configure_stream("EXTENDED_SYS_STATE", 1.2f);
		configure_stream("WIND_COV", 0.9f);
```
Working:
```
                configure_stream("SYS_STATUS", 1.0f);
		configure_stream("ATTITUDE", 15.0f);
		configure_stream("GPS_RAW_INT", 3.0f);
		configure_stream("GLOBAL_POSITION_INT", 3.0f);
		configure_stream("CAP_STATUS", 1.1f);
		configure_stream("VFR_HUD", 2.0f);
		configure_stream("EXTENDED_SYS_STATE", 1.0f);
		configure_stream("WIND_COV", 1.0f);
```

I added CAP_STATUS to common.xml with ID: 180 and generated mavlink 2.0 libraries. Mavlink versions match. 

I also realized that sometimes mavlink would stop for a second or two and then send all the missed messages really fast. Another case was, when I would receive no CAP_STATUS message (others were working) and all of a sudden a ton of them (50-100) would be sent in no time. Thus mavlink being unstable.

I know it is hard for others to debug this as the code is modified, but I think the bug is related to the source and can potentially affect other users.

bkueng, 2016-07-12:
There has been some recent changes to mavlink streaming (71d150f6ec0789d36d4f7af85a2e44c225846d51). Can you try with the latest master?

If it does not work, can you identify the exact commit (use git bisect) where it stopped working?

LorenzMeier, 2016-07-12:
The report references changes from quite a while back and with custom changes. Can you reproduce your issue on current master *without* any custom changes? Did you make sure your MAVLink app is not running out of flash space?

LorenzMeier, 2016-07-12:
I meant stack / RAM of course.

vvranjek, 2016-07-12:
Ok, so I managed to reproduce a similar bug on master (817f695), which is most likely related. Here is how to reproduce it:
```
git checkout 817f695
make clean
make submodulesclean
```

Comment the following drivers otherwise there is not enough flash
```
	##drivers/frsky_telemetry
	##drivers/px4flow
	##drivers/gimbal
```
The only changes I made is set all mavlink streams in mode 'Normal' to 1.0f as such:
```
configure_stream("SYS_STATUS", 1.0f);
		configure_stream("EXTENDED_SYS_STATE", 1.0f);
		configure_stream("HIGHRES_IMU", 1.0f);
		configure_stream("ATTITUDE", 1.0f);
		configure_stream("RC_CHANNELS", 1.0f);
		configure_stream("SERVO_OUTPUT_RAW_0", 1.0f);
		configure_stream("ALTITUDE", 1.0f);
		configure_stream("GPS_RAW_INT", 1.0f);
		configure_stream("ADSB_VEHICLE", 1.0f);
		configure_stream("DISTANCE_SENSOR", 1.0f);
		configure_stream("OPTICAL_FLOW_RAD", 1.0f);
		configure_stream("VISION_POSITION_NED", 1.0f);
		configure_stream("ESTIMATOR_STATUS", 1.0f);
		configure_stream("NAV_CONTROLLER_OUTPUT", 1.0f);
		configure_stream("GLOBAL_POSITION_INT", 1.0f);
		configure_stream("LOCAL_POSITION_NED", 1.0f);
		configure_stream("POSITION_TARGET_GLOBAL_INT", 1.0f);
		configure_stream("ATTITUDE_TARGET", 1.0f);
		configure_stream("HOME_POSITION", 1.0f);
		configure_stream("NAMED_VALUE_FLOAT", 1.0f);
		configure_stream("VFR_HUD", 1.0f);
		configure_stream("WIND_COV", 1.0f);
```
after that:
```
make
make -C build_px4fmu-v2_default upload
```
 - Take out SD card, just to make sure nothing is interfering
 - Open console on serial 5 and run
```
mavlink stop-all
mavlink start -d /dev/ttyS1 -b 115200 -m normal
```
 - Open qGCS and connect to this port.
 - Open MavLink Inspector

Now you should see that all messages received are at 0.5Hz (instead of 1.0Hz) except the heartbeat, which is at 1.0Hz. This isn't exactly the bug I originally posted, but is most likely related and should be reproducible.

LorenzMeier, 2016-07-12:
What you are observing is MAVLink's auto-throttling of the link bandwidth because the requested message rates are higher than the total link bandwidth available. So what you get is exactly what the system is designed to do.

vvranjek, 2016-07-13:
Hi @LorenzMeier,
The majority (2/3) of above messages aren't even sent because they aren't updated, which leads to 5 messages being sent at 0.5Hz is the limit of MAVLink @ 115200 baudrate? Isn't this preposterous? I understand the auto-throttling feature, but obviously 115200 is enough boudrate to send much more then 5, even 10 messages every 2 seconds (0.5Hz). I've been successfully  using MAVLink @ 115200 to send much more data then that before. Just ATTITUDE is typically sent at 15-20Hz which is already  more data then I am getting here..? 

Anyway, I was just trying to reproduce the bug on non-modified branch - master, but I guess this isn't it... I'll see what else I can do.

dagar, 2016-07-15:
@vvranjek can you share the output of `mavlink status` as this is happening. It should tell you the rate multiplier being applied to your streams. I'd need to take a closer look at the current implementation, but I believe the rate is scaled by the total size * rate of all specified streams, even if they aren't actively updating.

mhkabir, 2017-04-03:
Closing as stale. Potentially same issue : https://github.com/PX4/Firmware/issues/6235

closed 2017-04-03 by mhkabir