PX4/PX4-Autopilot #5029: pixhawk + px4 v1.4.1rc2 - unable to calibrate gyro and compass

svpcom, 2016-07-11:
When I use px4 v1.4.1rc2 with calibration from v1.3.4 or 1.4.1rc1 and doing rotate vehicle yaw half turn i've got *full* turn rotation in QGC.  When i've tried to calibrate gyro i've got `` Gyro #0 failure : Stale data!``.  Compass calibration also failed (see below).

HW: Pixhawk (fmu-v2) with hexa frame
FW: v1.4.1rc2 in EKF2 mode

With v1.3.4 or v1.4.1rc1 all works fine.

```
2016_07_10_22_59_13: [cal] calibration started: 2 gyro
2016_07_10_22_59_18: [cal] progress <25>
2016_07_10_22_59_19: [cal] progress <30>
2016_07_10_22_59_20: [cal] progress <35>
2016_07_10_22_59_21: [cal] progress <40>
2016_07_10_22_59_22: [cal] progress <45>
2016_07_10_22_59_23: [cal] progress <50>
2016_07_10_22_59_24: [cal] progress <55>
2016_07_10_22_59_25: [cal] progress <60>
2016_07_10_22_59_26: [cal] progress <65>
2016_07_10_22_59_27: [cal] progress <70>
2016_07_10_22_59_28: [cal] progress <75>
2016_07_10_22_59_29: [cal] progress <80>
2016_07_10_22_59_30: [cal] progress <85>
2016_07_10_22_59_31: [cal] progress <90>
2016_07_10_22_59_32: [cal] progress <95>
2016_07_10_22_59_33: [cal] progress <100>
2016_07_10_22_59_34: Gyro #0 failure : Stale data!
2016_07_10_22_59_35: [cal] calibration done: gyro
2016_07_10_23_00_01: [cal] calibration started: 2 mag
2016_07_10_23_00_03: [cal] Skipped scale calibration, sensor 1
2016_07_10_23_00_03: [cal] pending: back front left right up down
2016_07_10_23_00_03: [cal] hold vehicle still on a pending side
2016_07_10_23_00_05: [cal] detected rest position, hold still...
2016_07_10_23_00_05: [cal] down orientation detected
2016_07_10_23_00_05: [cal] down orientation detected
2016_07_10_23_00_05: [cal] Rotate vehicle around the detected orientat
2016_07_10_23_00_06: [cal] Continue rotation for down 7 s
2016_07_10_23_00_40: Failed: This calibration requires rotation.
2016_07_10_23_00_42: [cal] down side calibration: progress <4>
2016_07_10_23_00_47: [cal] calibration failed: mag
2016_07_10_23_00_48: PREFLIGHT FAIL: GYRO #1 SELFTEST FAILED
```

LorenzMeier, 2016-07-12:
The first issue is not a real issue because the gyro only timed out because of the calibration. And the 2nd issue was cause by you as you did not rotate during the mag cal.

LorenzMeier, 2016-07-12:
We still need to fix the timeout message obviously.

svpcom, 2016-07-12:
@LorenzMeier I've rotated them (about yaw axis) all time during calibration. It seem that it unable to detect rotation. There is very strange behavior - when I rotate vehicle to some angle \alpha QGC show me 2 * \alpha rotation and when I stop - it slowly aligns to \alpha.  This is not a HW issue because all works with previous PX4 versions.

svpcom, 2016-07-12:
@LorenzMeier I've also tested v1.4.1rc2 in LPE and INAV mode - the same calibration issues. The only one exception, that in LPE and INAV modes compass agree with vehicle orientation. It seems that EKF2 now has broken compass model.

LorenzMeier, 2016-07-12:
@svpcom Ok, that's super helpful feedback.

@bkueng This sounds terribly broken. Can you look tomorrow at this? Something is not consistent with the sensor sampling any more. Is the rotation param being applied incorrectly?

bkueng, 2016-07-13:
I havn't seen `Gyro #0 failure : Stale data!` so far, but I could reproduce and fix the 2 alpha problem.

svpcom, 2016-07-13:
What hardware do you use - pixhawk (fmu-v2)? I use pixhawk with external compass (mounted on gps module)

svpcom, 2016-07-13:
Also can you reproduce case when px4 unable to detect rotation (using external compass) during compass calibration?

bkueng, 2016-07-13:
@svpcom I used pixhawk v2 as well, no external devices attached.
If it's reproducible, can you use `git bisect` to find which commit broke it?

svpcom, 2016-07-13:
d846ad5dacfa4ab89fd611116d0ea8d21b9bc181

svpcom, 2016-07-13:
I've verified this on another pixhawk (from different supplier).
my config is:   hexa-x frame, EKF2. gcc 4.8.4
When all sensors calibrated recalibrate gyro and got this error.

julianoes, 2016-07-14:
Should be fixed now with #5062.

closed 2016-07-14 by julianoes