PX4/PX4-Autopilot #5037: PX4FMU driver: minor typo

akru, 2016-07-12:
I'm not sure but It's seems like a typo:
https://github.com/PX4/Firmware/blob/master/src/drivers/px4fmu/fmu.cpp#L1322

```c
...
		} else {
			// Scan the next protocol
			set_rc_scan_state(RC_SCAN_SUMD);
		}
...
```

in the case statement
https://github.com/PX4/Firmware/blob/master/src/drivers/px4fmu/fmu.cpp#L1288

```c
	case RC_SCAN_SUMD:
		if (_rc_scan_begin == 0) {
```

julianoes, 2016-07-13:
I don't understand. What would you change to what?

akru, 2016-07-13:
```diff
@@ -1319,10 +1319,9 @@ PX4FMU::cycle()
 
                } else {
                        // Scan the next protocol
-                       set_rc_scan_state(RC_SCAN_SUMD);
+                       set_rc_scan_state(RC_SCAN_PPM);
                }
 
-               set_rc_scan_state(RC_SCAN_PPM);
                break;
```

kd0aij, 2016-07-13:
That looks like a necessary fix. I guess SUMD has never been tested? Because it looks like the extraneous line 1325 would switch to PPM even if SUMD were detected.

akru, 2016-07-13:
https://github.com/PX4/Firmware/blob/master/src/drivers/px4fmu/fmu.cpp#L1325
looks like temporary hook for SUMD disabling. So SUMD correctly works now?

kd0aij, 2016-07-13:
@akru I'm unable to test that case, since I don't have any Graupner equipment.

julianoes, 2016-07-16:
@akru thanks for reporting! A fix has been merged.

closed 2016-07-16 by julianoes