PX4/PX4-Autopilot #5038: replay broken on master?

CarlOlsson, 2016-07-12:
Hi,

I can't run replay mode anymore on master. I get the error:

```: In member function ‘void Ekf2Replay::setEstimatorInput(uint8_t*, uint8_t)’:
/home/carl/Wingtra/Standard_PX4/Firmware/src/modules/ekf2_replay/ekf2_replay_main.cpp:446:7: error: ‘struct vision_position_estimate_s’ has no member named ‘timestamp_computer’
   _ev.timestamp_computer = replay_part5.time_ev_usec; // fake this timestamp
       ^
compilation terminated due to -Wfatal-errors.```

Is this because I have an old replay log file or is replay broken atm?

bkueng, 2016-07-12:
no it's really broken by 07384d6b5af38077299e

LorenzMeier, 2016-07-12:
Fixed in https://github.com/PX4/Firmware/commit/7419151314e7667d76a82565d3fd1c40bf416d2a

CarlOlsson, 2016-07-13:
Replay seems to still be broken. Now I do not get a warning message but the replay_replayed.px4log file is empty. I will look into it later

julianoes, 2016-07-13:
@CarlOlsson I could have broken it with #5003. Try before and after.

bkueng, 2016-07-13:
Also consider 9c73eae941c2ae569559878160ff66810161fad5: If you recorded before this and replay with a version after that it won't work.

julianoes, 2016-07-14:
@CarlOlsson can you please check this? There has also been a question on px4users about replay logging not working.

CarlOlsson, 2016-07-15:
Yes I can't replay my old replay logs with current master

JohnSnowball, 2016-07-15:
Hi, I got the same problem, I compile with
> make px4fmu-v2_ekf2

and reset the other params according to devguide．
btw, during the flight my 450 suffers a little bit oscilliation. 

Then for both raw log(before replay) and replay log file, I can't read them. I drag the file to flightplot it shows nothing, I tried log muncher, it returns "error 500 internal error, etc".

The log file is not empty, I suppose it has content but we just can't read it.
[image]

JohnSnowball, 2016-07-15:
additional, I just updated my firmware this Monday

bkueng, 2016-07-15:
@CarlOlsson do you really need to replay old logs?

@JohnSnowball That's a different problem, if you can't read the logs before replay. Try to start flightplot via console and then see if there's an error output when loading the file.

JohnSnowball, 2016-07-15:
I don't know how to start flightplot via console, is there any document? I'm on windows.

bkueng, 2016-07-15:
no not that I know of, but the steps are:
- open a console
- change into the directory with `flightplot.jar`
- start FlightPlot: `java -jar flightplot.jar`

CarlOlsson, 2016-07-15:
It is not crucial for me to replay old logs, most important is that I can create new ones

CarlOlsson, 2016-07-18:
WIth current master 9fcf121380f364e32c8d5b03e6b491cd0ba59afc it is not possible to record replay logs in SITL. MAVExplorer can't parse log file.
```
Failed to parse RPL1/<QQQQQffffffffff with len 72 (remaining 17123)
Skipped 72 bad bytes in log (40, 239, 147) remaining=17123
Failed to parse RPL1/<QQQQQffffffffff with len 72 (remaining 15634)
Skipped 72 bad bytes in log (251, 10, 148) remaining=15634
Failed to parse RPL1/<QQQQQffffffffff with len 72 (remaining 14608)
Skipped 72 bad bytes in log (44, 22, 148) remaining=14608
Failed to parse RPL1/<QQQQQffffffffff with len 72 (remaining 13928)
Skipped 72 bad bytes in log (244, 76, 148) remaining=13928
Failed to parse RPL1/<QQQQQffffffffff with len 72 (remaining 12528)
Skipped 72 bad bytes in log (19, 89, 148) remaining=12528
Failed to parse RPL1/<QQQQQffffffffff with len 72 (remaining 12277)
Skipped 72 bad bytes in log (144, 118, 148) remaining=12277
Failed to parse RPL1/<QQQQQffffffffff with len 72 (remaining 10818)
Skipped 72 bad bytes in log (20, 128, 148) remaining=10818
Failed to parse RPL1/<QQQQQffffffffff with len 72 (remaining 10173)
Skipped 72 bad bytes in log (174, 137, 148) remaining=10173
Failed to parse RPL1/<QQQQQffffffffff with len 72 (remaining 9032)
```

bkueng, 2016-07-18:
see #5090

CarlOlsson, 2016-07-18:
Thanks @bkueng replay should be working now

CarlOlsson, 2016-09-08:
It seems like replay is again broken on master.
I get the error:
```
[100%] Built target px4
args: /home/carl/Wingtra/Standard_PX4/Firmware/build_posix_sitl_replay/src/firmware/posix/px4 none replay none /home/carl/Wingtra/Standard_PX4/Firmware /home/carl/Wingtra/Standard_PX4/Firmware/build_posix_sitl_replay
SITL ARGS
sitl_bin: /home/carl/Wingtra/Standard_PX4/Firmware/build_posix_sitl_replay/src/firmware/posix/px4
label: none
debugger: replay
program: none
model: /home/carl/Wingtra/Standard_PX4/Firmware
src_path: /home/carl/Wingtra/Standard_PX4/Firmware/build_posix_sitl_replay
build_path:
cp: cannot stat ‘/home/carl/Wingtra/Standard_PX4/Firmware/build_posix_sitl_replay/Tools/posix_lldbinit’: No such file or directory
make[4]: *** [src/firmware/posix/CMakeFiles/replay] Error 1
make[3]: *** [src/firmware/posix/CMakeFiles/replay.dir/all] Error 2
make[2]: *** [src/firmware/posix/CMakeFiles/replay.dir/rule] Error 2
make[1]: *** [replay] Error 2
make: *** [posix_sitl_replay] Error 2
```
@LorenzMeier 
This commit broke it:
f511d49cc23898922d2169fa595258574f9b1dc6

CarlOlsson, 2016-10-03:
@LorenzMeier @priseborough Here is the old issue about replay not building

priseborough, 2016-10-03:
I worked through this with @tumbili tonight and it resolved when I updated my Xcode version on the Mac.

CarlOlsson, 2016-10-07:
Just tried it on Ubuntu and now it works (Y)

closed 2016-10-07 by CarlOlsson

shawinmihail, 2017-04-15:
I have the same problem - empty replay_replayed file, can someone help me?