PX4/PX4-Autopilot #5040: Airspeed validator reporting comm error

sanderux, 2016-07-12:
Airspeed sensor reading 0 (because there is no wind) results in a preflight check fail. (comm error)

tubeme, 2016-08-03:
@sanderux Look when I reported this :) https://github.com/PX4/Firmware/issues/4310 

Finally you got the cause. I wrongly accused the Eagle Tree Airspeed sensor.

Because the ET Airspeed V3 is very steady airspeed sensor when it is in the room it shows 0.0 m/s sharp when calibrated. So there is now way I could arm in the room. It gives AIRSPEED SENSOR COMM ERROR.

Now after you found the BUG i just have to blow in the pitot and voala it arms like a charm. 

Thanks man!

silkstone, 2016-08-25:
Hi @tubeme I am also getting AIRSPEED SENSOR COM ERROR with my digital airspeed sensor. Not sure if I am doing the calibration wrong. On QGroundControl it tells me to insure it is out of the wind and then it never promts me to blow across it although I am guessing that is what is needed for the calibration? Any help would be much appreciated!!

dagar, 2017-02-03:
I think this is resolved?

closed 2017-02-16 by dagar