PX4/PX4-Autopilot #5045: MC pos control: switching from MANUAL into ALTCTL

tumbili, 2016-07-13:


tumbili, 2016-07-13:
@kd0aij I'm talking about this here https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1593

kd0aij, 2016-07-13:
The current setting works well for me if MPC_THR_HOVER is set correctly, but I have no objection to changing it back if you feel the old method is more robust. I think all of my transitions from manual to altctl have been performed when hovering, not climbing or descending. But if the throttle stick isn't centered at transition, we can expect a change in vertical speed on transition anyway.

tumbili, 2016-07-13:
@kd0aij How about using the parameter if it's value is non-zero and having the default parameter value of zero?

kd0aij, 2016-07-13:
I'd avoid adding that complexity, and just change the one line of code. It should have no impact on people who switch from manual to altctl when hovering, and I doubt that I'd notice any difference if I were to change modes in a dynamic situation.

mhkabir, 2017-04-03:
This has been fixed in the recent controller refactor.

closed 2017-04-03 by mhkabir