PX4/PX4-Autopilot #5049: Quadcopter vision position estimation with INAV, OFFBOARD mode, shaking problem

rawk-v, 2016-07-13:
Hello, guys!
I am testing `OFFBOARD` mode in `INAV` estimator, with only **vision position estimation** used, no GPS, no FLOW.
I use `msf framework` and `mavros` to send vision data to Pixhawk. I have make the quadcopter takeoff in `OFFBOARD` mode and hold in a position. The only problem is the quadcopter **keep shaking** in position hold(a little heavily). 

Software information:
PX4, v1.3.3, with INAV estimator enabled;

This is my log information:[http://logs.uaventure.com/view/aVtvPo3PHXUfHby4FpApDV](http://logs.uaventure.com/view/aVtvPo3PHXUfHby4FpApDV)

And this is some `/mavros/local_position/pose` plotting data, I think the vision position estimate data is good enough.
[image]

I don't know if this is firmware side problem, can someone help? Thanks!

rawk-v, 2016-07-13:
Also, I have set the weight of vision position estimation to **10**, but the `local_position_ned/z` does not seem to as good as `/msf_core/pose/pose/pose/position/z`, as the above picture, `/msf_core/pose/pose/pose/position/z` is very precise, but `/mavros/local_position/pose/position/z`(`local_position_ned/z`) tend to be a slightly wrong as time elapse.

mhkabir, 2016-07-13:
@rawk-v Try LPE for a change, should improve your estimation quality. :)

rawk-v, 2016-07-14:
But it seems INAV has smaller error.
INAV estimator:
[image]
[image]

LPE estimator:
[image]
[image]

rawk-v, 2016-07-14:
LPE parameters:
```
x   LPE_INTEGRATE [69,155] : 1
x   LPE_FLW_OFF_Z [70,156] : 0.0000
x   LPE_FLW_XY [71,157] : 0.0100
x   LPE_FLW_QMIN [72,158] : 75
x   LPE_SNR_Z [73,159] : 0.0500
x   LPE_SNR_OFF_Z [74,160] : 0.0000
x   LPE_LDR_Z [75,161] : 0.0300
x   LPE_LDR_OFF_Z [76,162] : 0.0000
x   LPE_ACC_XY [77,163] : 0.0454
x   LPE_ACC_Z [78,164] : 0.0454
x   LPE_BAR_Z [79,165] : 1.0000
x + LPE_GPS_ON [80,166] : 0
x   LPE_GPS_DELAY [81,167] : 0.2500
x   LPE_GPS_XY [82,168] : 2.0000
x   LPE_GPS_Z [83,169] : 100.000000
x   LPE_GPS_VXY [84,170] : 0.2500
x   LPE_GPS_VZ [85,171] : 0.2500
x   LPE_EPH_MAX [86,172] : 3.0000
x   LPE_VIS_XY [87,173] : 0.5000
x + LPE_VIS_Z [88,174] : 0.5000
x   LPE_VIS_ON [89,175] : 1
x   LPE_VIC_P [90,176] : 0.0500
x   LPE_PN_P [91,177] : 0.1000
x   LPE_PN_V [92,178] : 0.1000
x   LPE_PN_B [93,179] : 0.0010
x   LPE_PN_T [94,180] : 0.0010
x   LPE_FGYRO_HP [95,181] : 0.1000
x   LPE_LAT [96,182] : 40.4300
x   LPE_LON [97,183] : -86.9290
```

INAV parameters:
```
    INAV_W_Z_BARO [-1,307] : 0.5000
    INAV_W_Z_GPS_P [-1,308] : 0.0050
    INAV_W_Z_GPS_V [-1,309] : 0.0000
  + INAV_W_Z_VIS_P [-1,310] : 10.00000
  + INAV_W_Z_LIDAR [-1,311] : 0.0000
    INAV_W_XY_GPS_P [-1,312] : 1.0000
    INAV_W_XY_GPS_V [-1,313] : 2.0000
    INAV_W_XY_VIS_P [-1,314] : 7.0000
    INAV_W_XY_VIS_V [-1,315] : 0.0000
  + INAV_W_MOC_P [-1,316] : 0.0000
    INAV_W_XY_FLOW [-1,317] : 0.8000
    INAV_W_XY_RES_V [-1,318] : 0.5000
    INAV_W_GPS_FLOW [-1,319] : 0.1000
    INAV_W_ACC_BIAS [-1,320] : 0.0500
    INAV_FLOW_K [-1,321] : 1.3500
    INAV_FLOW_Q_MIN [-1,322] : 0.3000
x   INAV_LIDAR_ERR [199,323] : 0.2000
    INAV_LAND_T [-1,324] : 3.0000
    INAV_LAND_DISP [-1,325] : 0.7000
    INAV_LAND_THR [-1,326] : 0.2000
    INAV_DELAY_GPS [-1,327] : 0.2000
    INAV_FLOW_DIST_X [-1,328] : 0.0000
    INAV_FLOW_DIST_Y [-1,329] : 0.0000
    INAV_DISAB_MOCAP [-1,330] : 0.0000
    INAV_LIDAR_EST [-1,331] : 0.0000
    INAV_LIDAR_OFF [-1,332] : 0.0000
```

@mhkabir  Is any weight params in LPE like `INAV_W_Z_VIS_P` INAV I can set?

mhkabir, 2016-07-14:
Yeah. Reduce the LPE_VIS_XY to the minimum value.

rawk-v, 2016-07-15:
Yes, thank you! Also set `LPE_VIS_Z` to minimum value, I can get pretty good result now.

mhkabir, 2016-07-15:
:)

closed 2016-07-15 by mhkabir