PX4/PX4-Autopilot #5060: Completely unstable MC flight v1.4.1rc2

ilia-sheremet, 2016-07-13:
I had successful flight with tag v1.4.1rc1 logs [here](http://logs.uaventure.com/view/DzRQoowFfGX5HzUae966nJ#_PLOT) (intensive wind conditions)

But when I switched to tag v1.4.1rc2 or master (at today's commit eda2915f0b4706ddfbd6df9770ce237baa5c5ad9) I have following problems:
- I experience the same issue as her #5029.
- It get GPS 3d fix but doesn't go ALT hold, unless you put LPE_GPS_ON = 0
- a copter is completely unstable - barely hold the plane attitude indoor and falls down after few seconds outside with wind. Estimator - LPE, all the PIDs are the same as the were for excellent performance flights. Some falling logs: http://logs.uaventure.com/view/EQAsb8tZDSbbsrKTMDhZzB
http://logs.uaventure.com/view/VPVTTbxNrf5TWXn7PLnHdD

julianoes, 2016-07-14:
Thanks for reporting. I'm assuming the error was due to #5029 and fixed with #5047. @bkueng am I right?

bkueng, 2016-07-14:
@julianoes yes the stability issues should be gone. Some more testing would be welcome, just to be sure. There's still the calibration issue `Gyro #0 failure : Stale data!`. Could you look into that? It seems that after calibration, the gyro 0 doesn't return any data for a while and thus the voter fails over to the next gyro with a timeout message.

julianoes, 2016-07-14:
@bkueng strange. I've seen similar issues on Snappy before where I would see baro timeout from the commander (which hints at a slow sensor update) right at the end of the calibration.

ecmnet, 2016-07-14:
@bkueng, @julianoes  In addition to the message `Gyro #0 failure : Stale data!`, I also get a message, that in preflight checks Gyro0 failed. Does Gyro calibration work at all? With latest master, I get I continuously increasing pitch offset, which is hardly to compensate by RC after a couple of seconds. Recalibration did not fix this.

julianoes, 2016-07-14:
@ecmnet Hm yes, something seems wrong. I'm looking into it as well.

ilia-sheremet, 2016-07-17:
Thank's! Will do more tests on Monday!

ecmnet, 2016-07-18:
@julianoes I did some more flight tests with the latest master. Still I get an increasing pitch drift with it. Reverting to 1.3.4 solves that issue.

julianoes, 2016-07-18:
@ecmnet: can you share a log where that happens?

ecmnet, 2016-07-18:
@julianoes Here is one: http://logs.uaventure.com/view/GwPvYyz3iR5vEsiEpoVmPh

julianoes, 2016-07-19:
@ecmnet I had a look at the log but I don't know how I could see or tell the pitch drift.
Does it also happen when you're not flying but the quad is just sitting on the table? If yes, can you try to arm (e.g. without props) and without auto-disarm set and leave it for some minutes?

ecmnet, 2016-07-19:
@julianoes If you compare  http://logs.uaventure.com/view/GwPvYyz3iR5vEsiEpoVmPh#AS_Pitch_PLOT with the radio input shown in http://logs.uaventure.com/view/GwPvYyz3iR5vEsiEpoVmPh#RC_Raw_PLOT (blue line for pitch) you can see that i had to correct pitch massively to keep the vehicle at its position. It as also very obvious, that the effect was increasing over the time. After a couple of seconds I was hardly able to control the vehicle at all.

Currently I have 1.3.4 on my quad which works fine. At the end of the week i could be able to test master again.

ilia-sheremet, 2016-07-20:
I did tests yesterday with v1.4.1rc1 and v1.4.1rc2 master. The initial issue was solved at master so the copter was capable to flight, but the performance is significantly worse than v1.4.1rc1.
Here are the logs of the same day with the same params:
v1.4.1rc1 - http://logs.uaventure.com/view/Y8gCg2BVo6bkpwGermLLcN
v1.4.1rc2 master - http://logs.uaventure.com/view/Dv85xGWkRmqnaeyQptb33Z

On the latter one the copter is much more twitchy, holding the XYZ position much worse.
I am also getting INFO messages like these:
**2016_07_19_15_42_13: [lpe] gps fault   2   0  23   0   0   0
2016_07_19_15_42_17: [lpe] gps fault   2   0  15   0   0   0**
flying rc1 firmware I might get 2-3 of these messages during the 15 min flight, while flying rc2 I get around 100 of these messages.

It doesn't look like a hardware issue, since I tried to change GPS modules to different vendors and the issue remains.

ecmnet, 2016-07-20:
The pitch issue might be connected with #4879.

julianoes, 2016-07-20:
> The pitch issue might be connected with #4879.

I don't think it's connected to the integral gain of the controller because the control performance looks decent looking at ATT.Pitch vs. ATSP.PitchSP.

julianoes, 2016-07-20:
@ilia-sheremet Thanks for reporting. We'll have to do some testing to try to reproduce your issues.

ecmnet, 2016-07-20:
@julianoes You are right, can't be the integrator. Comparing the Gyro data between 1.3.4 and master, I see a significant increase on noise. (1.3.4 <+/-0.05 vs. master <+/-0.3). Any idea where it could come from?

ecmnet, 2016-07-22:
@julianoes I made some comparison with the quad on the ground. Procedure: arming and then increasing throttle (just not taking off).

GyroY 1.3.4 =>

[image]

GyroY Master 8ab841e0467d8509db9f97019c640ea8da2cdaff =>

[image]

Pitch 1.3.4 =>

[image]

Pitch Master 8ab841e0467d8509db9f97019c640ea8da2cdaff =>

[image]

There seems to be a significant difference in scale.

julianoes, 2016-07-22:
@ecmnet Interesting, thanks for the data! Can you tell me what Hardware you are using? Pixhwak, Pixfalcon, Pixracer? And by any chance, do you know which gyro is used, either the MPU6000 or L3DG20.

You could also compare the values doing:

```
mpu6000 test
l3gd20 test
```

ecmnet, 2016-07-22:
@julianoes I am using Pixhawk. Both running with standard setup (LPE). Without arming, I get

1.3.4 =>

mpu6000: gyro y:          0.00346       rad/s
l3gd20: gyro y:               0.00255       rad/s

Master =>

WARN  [mpu6000] gyro y:           0.00267       rad/s
WARN  [l3gd20] gyro y:              -0.00223       rad/s

Besides the sign for L3GD20 in master, quite similar.

How can I figure out which gyro is used? The data shown above is just MAVLink published data.

julianoes, 2016-07-22:
@ecmnet I think you're just looking at noise if you just look at one instant.

If you're on a Pixhawk and you're not getting any messages like "gyro 0 fail, failover", it should be using 0 which is the one that is started first in https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rc.sensors#L94, so the MPU6000.

Let me try to reproduce what you found.

julianoes, 2016-07-22:
@ecmnet I compared the two versions on my Pixhawk and I can't see a difference, both gyro calibrations work as expected and the attitude estimate is decent.

If you see a very clear difference, could you try to do a git bisect between the two versions to see which commit causes the troubles?

ecmnet, 2016-07-22:
@julianoes The difference in attitude estimation and gyro is cleary visible above. I'll try to figure out, which commit seems to be responsible for this effect. I understood, that you were not be able to reproduce my issue. Did you do the kind of throttle-ramp-up, I did? The strange thing is, that when running the ash tests, the magnitude of 1.3.4 and master is the same. Looking at the published MAVLink-Data - even when not armed -, there seems to be a difference by nearly half an magnitude (1.3.4: about 0.02 vs. master about 0.07), if armed this get worse.

tumbili, 2016-07-22:
@ecmnet I just have some general comments for you not comparing different versions.
In most of the logs I have seen above the IMU data is all over the place, indicating that your vibration damping is not optimal. Particularly here you can see that LPE diverged and did a reset http://logs.uaventure.com/logmuncher/charts/GwPvYyz3iR5vEsiEpoVmPh/Local_X.png
Additionally, the attitude estimator q uses the estimated velocities form the state estimator to compensate for non-gravitational acceleration. So if your velocity estimation diverges or is bad I can well imagine that your attitude will also suffer from this.
I think you need to start fixing your setup!

ecmnet, 2016-07-22:
@tumbili I totally agree with you and I am aware of the vibrations already.  
Nevertheless, I am curious about the different behavior between versions, which I try to understand. 
So my basic question is, what has changed. LPOS, in my case, is based on PX4Flow data only and has continuously timeouts, etc - those should - in my opinion - not affect the attitude at all. Was the compensation of acceleration based on the state estimations introduced shortly?

tumbili, 2016-07-22:
@ecmnet I'm not exactly sure what's the cause of it but there are several possibilities.
Having a brief look at the last logs from @ilia-sheremet I just notices than in his old logs the range of acceleration seems less. I'm also not sure if between those versions something in the IMU driver code changed (e.g. filtering). Also changes to the estimator parameters could cause this, you would have to sync up on this with @jgoppert. If the LPE would log estimator innovations then that would make our life a bit easier analyzing the issue, see https://github.com/PX4/Firmware/issues/4762

jgoppert, 2016-07-22:
@tumbili looking at the ekf2_innovations message. What is vel_pos_innov? Is that just for gps measurement?

tumbili, 2016-07-22:
@jgoppert those are generic innovations for NED velocity and position. At the moment ekf2 does not fuse several measurement sources for the same axis.
So e.g. if you are using baro to estimate height, then vel_pos_innov[5] will be the baro innovation, however, if you are using a range finder for height estimation then it will be the innovation for the range finder.

jgoppert, 2016-07-22:
Just created a PR #5124 to add innovation logging to LPE, so this may help diagnose this.

ecmnet, 2016-07-22:
@jgoppert @tumbili @julianoes This might help during flight, but I doubt, that this would help to explain the effects on ground, I've shown above. Meanwhile I did some simple vibration damping, and was able to decrease gyro noise by about 30%. Again I did a on-ground-comparison between 1.3.4 and master - and again, I saw pitch increasing over the time on ground. Flight shows again the effect: Only slow drift with 1.3.4 and again increasing pitch drift with master.

ecmnet, 2016-07-22:
@julianoes Just tested 1.4.1 RC1. This version works fine for me.

tumbili, 2016-07-22:
@ecmnet Please share the latest log where pitch drifts

ecmnet, 2016-07-22:
@tumbili I posted one 4 days ago: : http://logs.uaventure.com/view/GwPvYyz3iR5vEsiEpoVmPh

tumbili, 2016-07-22:
@ecmnet Which one is the one on the ground?

ecmnet, 2016-07-22:
@tumbili For the ground test, I've no logs, but published the screenshots above.

tumbili, 2016-07-22:
@ecmnet I need those logs :-)

ecmnet, 2016-07-22:
@tumbili Try to find them...did a lot tests today

tumbili, 2016-07-22:
@ecmnet In those plots were you actually moving the quad or was it just sitting on the ground?

ilia-sheremet, 2016-07-22:
@jgoppert does your PR suppose to solve the issue I mentioned regarding **[lpe] gps fault** ?

tumbili, 2016-07-22:
@ilia-sheremet No, it's only to help us analyze the problem.

tumbili, 2016-07-22:
@ecmnet I also see that your max gyro bias for the attitude estimator is set to zero.

ecmnet, 2016-07-22:
@tumbili With RC2 the drift appears again. Here's the log while on the ground, arming and increasing throttle: http://logs.uaventure.com/view/kARbQ4oPptvvpSX3nzz9oS

ecmnet, 2016-07-22:
@tumbili Is that a new parameter? Does that mean, that gyro bias is not compensated?

tumbili, 2016-07-22:
@ecmnet No it's not new at all. And yes, I guess that's what it means.

ilia-sheremet, 2016-07-22:
@tumbili Will make another log then with this PR.

ecmnet, 2016-07-22:
@tumbili Same test on ground again with 1.4.1RC1 for comparison: http://logs.uaventure.com/view/wuC4PJ5CnvVKKHaW9nGmi4

tumbili, 2016-07-22:
@ecmnet I compared the two logs and performed a manual integration of the rates to see if they correlate with the estimated roll and pitch angles. It really seems like the older version gets you better results. I scrolled through the commits and the biggest changes between the two versions is the refactoring of the sensor topics @bkueng FYI

ecmnet, 2016-07-22:
@tumbili thanks a lot for your effort analyzing the data. Do you have some measurements/logs to benchmark the setup?

bkueng, 2016-07-22:
A possible cause of sensor value changes that I see with my sensors refactoring is: the raw sensor topic has both the values & the integral. After the refactoring, if the integral_dt is set, the integral value will be used (prefered choice), otherwise the value is used. So you can check what has been used before (I think it was the integral as well), and whether using the values improves things.

tumbili, 2016-07-22:
@bkueng I believe that before the values were used not the integral  but I need to check. In theory it should result in the same thing but who knows...

ilia-sheremet, 2016-07-22:
@jgoppert Just made a log with your [PR](https://github.com/PX4/Firmware/pull/5124). 
Here is the link to [uaventure log](http://logs.uaventure.com/view/reE9Y7Ladzmm7hvUF6GjGi), or the [px4log](https://drive.google.com/open?id=0B4HSR_JQFK8Ka2pLVV9pUnBuZkk) itself and [message](https://drive.google.com/open?id=0B4HSR_JQFK8Kdnc3U3dKZ3JXQTQ) files. Apart from GPS there was also a baro fault message.

There was quite intensive wind around 8m/s with 10-11 gusts, the copter is also small around 500 class.

ecmnet, 2016-08-03:
@tumbili @bkueng The pitch issue appears also in V1.4.1RC4.

julianoes, 2016-08-03:
@ecmnet do you have a logfile for the last test? Is this with Pixracer or Pixhawk? How is it dampened?

ecmnet, 2016-08-03:
@julianoes Please find the log attached. This is with Pixhawk. I am already aware that damping is not optimal with my setup, but this issue occurs only from 1.4.1 RC2 on; RC1 works fine for me.

RC4 log: http://logs.uaventure.com/view/YwZWjefMWDd56KaLW6iBy8
RC1 log:  http://logs.uaventure.com/view/w6HFPnTK8MkeQA3HJYg6J5

As mentioned by @bkueng the sensor logic was changed between those RCs.

bkueng, 2016-08-03:
@ecmnet Thanks for your tests. Just to be sure it's the sensors logic, can you change the line
https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensors.cpp#L1161
into:
```
	if (false) {
```
And the same for the `accel_poll` method?

ecmnet, 2016-08-03:
@bkueng @julianoes I just did some short tests changing sensors.cpp not to use the integrals.
1. Test: Changing gyro polling only: Seems to be better, but still some drift.
2. Test: Changing both, gyro and acc: The issue is fixed. Quite stable flight without any significant drift.

ecmnet, 2016-08-03:
@julianoes I'll do a test certainly. Perhaps you could give me some short explanation why this could be connected to the integration issue of sensor.cpp? As far as I can see you added a low pass filter to the attitude estimator. This would apply to both methods of sensor.cpp, doesn't it?

LorenzMeier, 2016-08-04:
The nature of the complimentary filter tuning is that it expects a certain conditioning of the input data and the frequencies contained in it. We achieve this with the filter.

ecmnet, 2016-08-04:
I can't see the addition of a complimentary filter in this change, but a low pass filter for acc and gyro data: https://github.com/PX4/Firmware/pull/5216/commits/1c683ea13556ce9e92a308533473e77c7344c0d3
As I faced a kind of bias as a result of the attitude estimation if using the gyro/acc integral divided by integral_dt instead of the reported value itself , my question is, how a low pass filter can help here, if it is applied in a later step. Anyway, I'll try to give it a test today.

LorenzMeier, 2016-08-04:
You need to understand the frequency response of attitude_estimator_q, which is a complimentary filter, to understand this change.

julianoes, 2016-08-04:
@ecmnet before the change in sensors.cpp, the filtered gyro/accel values were fed to the attitude estimator q, however with the change we're trying to move to the unfiltered but integrated sensor values.

[10 more replies]

PR #5062 (2016-07-14): Fix gyro calibration
The gyro calibration scale was initialized and set to 0 in a calibration instead of 1.

Tested using Pixhawk.

This fixes #5060 and #5029.

@bkueng please review and merge as soon as travis is ok.
closed 2016-08-31 by julianoes