PX4/PX4-Autopilot #5064: Issues relating PX4Flow sensor with PIXHAWK on QGC

siddhu45, 2016-07-14:
Hello community,
I've been trying to use the PX4Flow sensor (with sonar) on my PIXHAWK to implement Position Control using MAVROS. I setup everything as mentioned on the wiki pages and a few documentations I found. I still have some doubts:

1. I am able to use the Optical Flow sensor but unable to connect or use the sonar onboard PX4Flow sensor. I have connected the PX4Flow sensor with the Magnetometer via hub to I2C and the sonar to ADC 6.6V on Pixhawk. It doesn't seem to me that the sonar is giving any readable data. I tried a couple of things. First, I tried using the APM Firmware on the Pixhawk and connecting it to MP as it has direct sonar (Range finder) parameter (I don't know what parameters to change in order to use sonar in QGC, any insight on this too would be appreciated). This gave me no readings in the status tab in Flight data. I also tried connecting the same to APM 2.6 but same, it lights up, but does not give me any readings. Also, I tried connecting Pixhawk with MAVROS, but still no values turned up. So, any help on how to use the sonar on PX4Flow!

2. I want to use the ekf2 estimator for position control. So, how can I make sure that the ekf2 estimator uses the Optical Flow sensor data for sensor fusion. I mean, does it by default uses PX4Flow data for sensor fusion or do I need to change something else? After reading on other posts, I found that all I need to change the vale of ekf2_aid_mask parameter to 1 (in order to use the flow sensor data). Is that all?

3. Does the different estimators, INAV or LPE or EKF2, by default uses all the available sensors for sensor fusion (e.g. just as I connect PX4Flow, is its measurements are incorporated for sensor fusion or do we need to change certain parameter).

4. I tried to do Position Control with MAVROS (without PX4Flow) but the UAV seems to have very wrong estimate (or measurements) for Altitude. So, could anyone kindly confirm is the Pixhawk's position controller taking sensor fused data (combination of GPS + IMU + Barometer) coming from the chosen estimator or it is directly taking the data from barometer only? In a nutshell I mean, could anyone please confirm that Pixhawk's position control is based on sensor fused data or some individual sensor.

Thanks to all for contributing beforehand. Any help is appreciated.

Thanks & Regards
Siddharth

kd0aij, 2016-07-14:
Perhaps this guide will help: http://dev.px4.io/flow_lidar_setup.html

mhkabir, 2016-07-14:
> sonar to ADC 6.6V on Pixhawk.

We don't support that by default. Use the sonar on the flow board directly. Should be plug and play with the LPE estimator - http://dev.px4.io/flow_lidar_setup.html

siddhu45, 2016-07-15:
Thank you so much for a spontaneous response @kd0aij and @mhkabir. But I have already tried using the instructions on that page and still it doesn't seem to show me any reading. It powers up no doubt but I fail to receive any data. If you say its plug n play then if I select my estimator as LPE, should it work? 
Thanks a lot!

Regards
Siddharth

kd0aij, 2016-07-15:
Maybe it would help to post a log.

siddhu45, 2016-07-19:
Hi, Sorry for late response!
I connected the sonar just as on the wiki page of Lidar lite, but instead of 4 connections I only made 3 wire connections. I didnt know where to connect the PWR EN cable. So I connected the +V and GND to power it and the PWM (pin 2) on sonar to the Aux 5 as documented. I hope this connection is correct. The log is here:
```
pwm_input start
ERROR [pwm_input] driver already started
nsh> pwm_input test
Showing samples for 5 seconds
nsh> ll40ls start pwm
ERROR [ll40ls] driver already started
nsh> ll40ls test pwm
ERROR [ll40ls] immediate read failed
ERROR [ll40ls] Task exited with errno=22

nsh> ll40ls info pwm
state @ 10007f30
ll40ls_pwm_read: 17867 events, 0 overruns, 971172us elapsed, 54us avg, min 28us max 2318us 122.076us rms
ll40ls_pwm_read_errors: 17863 events
ll40ls_pwm_buffer_overflows: 0 events
ll40ls_pwm_zero_resets: 4 events
WARN  [ll40ls] poll interval:  50 ticks
WARN  [ll40ls] distance: 0.000m
nsh> 
```
I exactly did it step by step with just the one alteration as mentioned above. Please comment on this.

Thank you so much
Siddharth

closed 2017-04-03 by mhkabir