PX4/PX4-Autopilot #5070: Sensors can not be calibrated on MindPX

dengzhijie, 2016-07-15:
We found a problem in the v1.4.1rc1/v1.4.1rc2 on MindPX. Sensors could not be calibrated,when I clicked the sensor's name and choose OK to calibrate,it did not show any  information.But we could calibrate  without any problem in v1.4.0 and before. It seems to be influenced by recent changes,and we are checking the code.

bkueng, 2016-07-15:
Thanks for reporting. Can you try latest master? There were some fixes regarding calibration in the last few days.

airmind, 2016-07-17:
@bkueng The latest master has been tested on MindPX, and could not resolve this issue. @dengzhijie did you try different QGC version?

dengzhijie, 2016-07-18:
I've tryed the laeted master on MindPX,and it did not work.I also tryed different QGC versions on diffierent PC,but sensors still could not be calibrated.

bkueng, 2016-07-18:
Sorry, I can't reproduce, as I don't have a MindPX. If you say it works with v1.4.0, can you do a `git bisect` to find where it stopped working?

dengzhijie, 2016-07-22:
@bkueng We checked the code,and resolved the problem.Due to MAVLINK app needs more FDs,and we did not update the code timely, the problem appeared.
Really thanks to your help!

bkueng, 2016-07-22:
Glad to read that it works now!

closed 2016-07-22 by bkueng