PX4/PX4-Autopilot #5072: Unstable position hold using px4flow with pixhawk and QGC

Graceson, 2016-07-15:
Hello everyone,

I am having trouble to get a stable hovering using optical flow. I use a dji flamewheel F450 drone with pixhawk and QGC latest stable version. I don't have gps and I just use px4flow for position estimation. I have set the parameters to use the INAV position estimation method. When I switch to position control flight mode, the quad does a somewhat OK stable for 5 - 10 seconds, then it starts to drift. The altitude seems to be fairly constant, but the drifting is more in xy direction. Now when I try to get back to the position using the radio sticks (in position control mode itself) the quad goes really unstable and I have to switch it back to stabilize mode to regain control. I have tried the lpe position estimation mode also, but it is far more unstable than the INAV method. Is there any parameter change or setup that I can do to get more stable hovering using optical flow? I am happy since the quad is doing something when switched to position control (I took four weeks to get to this point). But I would like to get a better and reliable position hold. Since I am novice in this field, my apologies if my question is too basic. My project is a small part of a bigger research project, so I am in desperate need to get it working as soon as possible. Thanks in advance for everyone's help.
Also, below are the parameters than I changed to make the position control in INAV mode work: 

- SYS_MC_EST_GROUP -> position_estimator_inav, a Set multicopter estimator group

- INAV_LIDAR_EST -> 1 (I dont use a lidar, but this parameter change was needed to make position               control work otherwise it would give me a reject position control even if the OF data was available)

Thanks again!

julianoes, 2016-07-16:
> INAV_LIDAR_EST -> 1 (I dont use a lidar, but this parameter change was needed to make position control work otherwise it would give me a reject position control even if the OF data was available)

This seems suspicous and wrong. Can you retry current master after this fix please: #5058.

mhkabir, 2016-07-16:
It's easier to switch to LPE. Everything should work out of the box

mhkabir, 2016-07-16:
Set MC_EST_GROUP to Local Position Estimator. Let us know how it goes.

Graceson, 2016-07-16:
Thank you all for the suggestions. I will try the suggestion of Julianoes. And let you know the result. mhkabir, I have tried the LPE  method, but it was far more unstable than the inav method. It did work for few seconds, and then it started oscillating vigorously in the ZY plane and crashed

Graceson, 2016-07-16:
mhkabir, do you have any suggestions how to solve that oscillating motion in ZY  plane? Any parameters for tuning or something?

Graceson, 2016-07-16:
Sorry double post

Graceson, 2016-07-20:
And yes, we fixed the problem! We switched to the LPE position mode and then did some PID tuning for the following parameter : MPC_XY_P,MPC_XY_VEL_D, MPC_XY_VEL_I, MPC_XY_VEL_P, MPc_Z_VEL_I . Then the quad became pretty stable, and we are satisfied with hovering.

Graceson, 2016-07-20:
We also got stable hovering with the INAV method after tuning the MPC_XY_P parameter. But LPE is mich more better. Thank you @julianoes and @mhkabir

Graceson, 2016-07-20:
Now we have good amount of vibrations. We want to reduce the vibrations, so that we can take good pictures. So I am looking for props with smaller pitch ( currently we have a huge prop with pitch of 4.5 units, I believe a smaller prop will reduce the vibrations and give a smoother flight). I am also looking for the possibility of a gimbal. I have a Canon Powershot ELPH 130 IS.  Does anyone have a light weight gimbal suggestion? If not this issue can be closed. Again thank you very much!

mhkabir, 2016-07-20:
Please ask other questions on the PX4 Discuss.

closed 2016-07-20 by mhkabir

rhamrah, 2018-10-18:
> And yes, we fixed the problem! We switched to the LPE position mode and then did some PID tuning for the following parameter : MPC_XY_P,MPC_XY_VEL_D, MPC_XY_VEL_I, MPC_XY_VEL_P, MPc_Z_VEL_I . Then the quad became pretty stable, and we are satisfied with hovering.

How do you use Position mode without using GPS? In https://docs.px4.io/en/flight_modes/position_mc.html they say this mode requires GPS. I am using Px4flow sensor for x and y and tfmini for z direction without any GPS and EKF estimator. In Altitude mode, the drone maintains the altitude but drifts in x and y directions. Do you think Position mode will take care of these x-y drifts without using GPS?

julianoes, 2018-10-19:
This thread is closed. @rhamrah please create a new issue for your question.