PX4/PX4-Autopilot #5076: Small bug in recovery after mission download fails with a timeout.

lovettchris, 2016-07-15:
If a mission download fails with timeout the MavlinkMissionManager gets stuck thinking a transfer is in progress and therefore refuses to receive any more missions after that point.  I think the fix is to add the following line:

    	_transfer_in_progress = false;

inside the "check for timed-out operations" where it sends the following status:

    _mavlink->send_statustext_critical("Operation timeout");

Notice here it sets the _state to MAVLINK_WPM_STATE_IDLE, but it forgot to reset _transfer_in_progress and that causes handle_mission_count to return MAV_MISSION_ERROR from here on out.

LorenzMeier, 2016-07-15:
Could you maybe send a diff for the fix?

lovettchris, 2016-07-15:
Sure :-)

dagar, 2016-07-15:
This sounds pretty close to a bug I just saw today. I'll test the pull request.

lovettchris, 2016-07-15:
See  https://github.com/PX4/Firmware/pull/5077

dagar, 2016-07-16:
I've been doing a lot of mission testing that's included #5077. So far so good. Thanks @lovettchris

closed 2016-07-16 by dagar