PX4/PX4-Autopilot #5078: setting active waypoint and mission feasibility checker

dagar, 2016-07-16:
If you change the active waypoint (MISSION_SET_CURRENT) the mission feasibility will check your current distance from the first waypoint and possibly refuse the mission.

[image]

LorenzMeier, 2016-08-07:
Can you re-test on master? We should have fixed that, but not sure.

dagar, 2016-08-07:
@LorenzMeier have you tried gazebo plane recently?

My gazebo seems to be borken and I can't takeoff. Possibly because of this? I've completely reset to master and cleaned everything.

    [Err] [Plugin.hh:165] Failed to load plugin libLiftDragPlugin.so: libLiftDragPlugin.so: cannot open shared object file: No such file or directory
    [Err] [Plugin.hh:165] Failed to load plugin libLiftDragPlugin.so: libLiftDragPlugin.so: cannot open shared object file: No such file or directory
    [Err] [Plugin.hh:165] Failed to load plugin libLiftDragPlugin.so: libLiftDragPlugin.so: cannot open shared object file: No such file or directory
    [Err] [Plugin.hh:165] Failed to load plugin libLiftDragPlugin.so: libLiftDragPlugin.so: cannot open shared object file: No such file or directory

LorenzMeier, 2016-08-07:
Check the CMakeLists.txt in src/Tools/sitl_gazebo. It excludes the plugin if you're using a Gazebo version earlier than 7.0. We might need to either fix the plugin dir of Gazebo or just unconditionally re-enable this.

dagar, 2016-08-09:
Temporarily fixed by removing the version check here - https://github.com/PX4/sitl_gazebo/blob/master/CMakeLists.txt#L192

dagar, 2016-08-10:
@LorenzMeier @jgoppert what's the proper fix for gazebo plane? Is the version check backwards? Only enable the LiftDragPlugin plugin for gazebo >= 7?

LorenzMeier, 2016-08-10:
I'd say we always enable. Some LINUX distributions are missing the proper plugins (or the path setup for them).

dagar, 2016-08-16:
No longer a problem on current master.

closed 2016-08-16 by dagar