PX4/PX4-Autopilot #5080: Snapdragon: get HITL working

julianoes, 2016-07-16:
It would be nice to have HITL working again with Snapdragon.

FYI: @bharath374

mcharleb, 2017-04-13:
@julianoes This is not something we will be able to support on Eagle. Please close unless there is someone else who you think who could implement this.

julianoes, 2017-04-13:
HITL actually works well again on Pixracer/Pixhawk, Raspberry Pi, so it should be do-able on Snapdragon. One catch is that it requires high bandwidth. For the Raspberry Pi a LAN cable can be used, for the Snapdragon that's not an option, so we'd have to check the wifi bandwidth.

closed 2018-02-21 by PX4BuildBot

TSC21, 2018-02-21:
@julianoes @ChristophTobler does HITL got back to work with SDFlight?

julianoes, 2018-02-22:
In theory it could be tried. It might be tricky due to latency through wifi though.